don/cordova-plugin-ble-central #768: ios Problem - ble.isEnabled()

stefanotauriello, 2020-05-29:
It never gives Bluetooth enabled, even if I've activated from my phone

mohit-testframework, 2020-11-27:
Hello,

I also faced this issue at my side, Anyone may help me with this?

Thanks

peitschie, 2022-01-24:
Are you able to provide samples of how you're calling `ble.isEnabled`? This is working for me on a variety of projects that run under iOS, so I'm not clear why it's misbehaving for you.

tigohenryschultz, 2022-02-17:
Success or failure never triggers for iOS

```
  isBleEnabled: ({commit, dispatch}) => {
    log.debug('TigoBle2:isBleEnabled');

    return new Promise((resolve, reject) => {
      window.ble.isEnabled(
        () => {
          commit('setIsBleEnabled', true);
          resolve(true);
        },
        function () {
          commit('setIsBleEnabled', false);

          dispatch('enableBle')
            .then(resolve)
            .catch(reject);
        }
      );

    });
  },
```

peitschie, 2022-02-21:
Are you using `IOS_INIT_ON_LOAD=false` on this project, by chance?

peitschie, 2022-02-21:
I've just confirmed that this works locally regardless of the load setting I've noted above.
[image]

@tigohenryschultz have you tried dropping a breakpoint within those functions there to see if they trigger?

Are you using the ionic wrapper by any chance? As for the ionic wrapper, I suspect the syntax there would be different...

tigohenryschultz, 2022-02-22:
I breakpointed them in Android/iOS and have device logs for both callbacks, Android is working excellent as did our old implementation with (https://github.com/randdusing/cordova-plugin-bluetoothle) so referencing in the global scope has never been an issue for us. I'll try to explain the code flow in the other ticket sometime today.

peitschie, 2022-02-23:
Are you able to load your app up in Xcode and drop breakpoints in the native code to see if those are being hit?

Alternatively, can you put together a basic app I can get access to reproduce locally?

Are you using cordova, ionic or capacitor for this app?

tigohenryschultz, 2022-02-23:
I will see what I can dig up in Xcode today, we are using Ionic with the cordova plugin 'cordova-plugin-bluetoothle 6.5.1'.

Putting together a sample might be cumbersome right now, we are still using randusing until the iOS problems are solved with this one.

nunoarruda, 2022-02-23:
@peitschie The problem is that, even with BLE enabled + BLE access allowed for the app, isEnabled() will fail on the first call and the error will return `0`. If you call isEnabled() again it will work.

You can reproduce this by cold-starting your app (remove the app from the background first and then open the app) and call isEnabled().

The error returning `0` looks to be unexpected. Here's what I found about the errors a while ago when I extensively investigated this issue:
```
/**
 * iOS
 * 3 = BLE access not allowed
 * 4 = BLE functionality disabled
 *
 * BLE disabled & no BLE access = 3
 * BLE enabled & no BLE access = 3
 * BLE disabled & BLE access = 4
 *
 * Sometimes 0 is reported, possible bug? Toggling BLE functionality looks to be a workaround
 *
 *
 * Android
 * BLE disabled & no location permission = "Bluetooth is disabled."
 * BLE enabled & no location permission = fails on the 'scan' method with "Location permission not granted."
 * BLE disabled & location permission = "Bluetooth is disabled."
 */
```

tigohenryschultz, 2022-02-23:
@peitschie 
Instead of debugging in xcode I am going to create you guys a sample Ionic/Cordova app you can add to this repo for testing/troubleshooting.

peitschie, 2022-02-23:
@nunoarruda those symptoms sound identical to #828

The 0 code you're receiving is the `CBManagerState` reported by iOS
```
// Sourced from
// https://github.com/xybp888/iOS-SDKs/blob/6ba2e4fccdef4176eecf49e70fa66526c6b53ee2/iPhoneOS12.4.sdk/System/Library/Frameworks/CoreBluetooth.framework/Headers/CBManager.h#L31
export enum CBManagerState {
  Unknown = 0,
  Resetting,
  Unsupported,
  Unauthorized,
  PoweredOff,
  PoweredOn,
}
```

On iOS, the BLE adapter (even when authorized) takes a few moments to "warm up" before the application is allowed to use the adapter. A few attempts have been made a few years back to improve the behaviour, but no reliable fix has been found for the native code here.

The general approaches I use (depending on the application requirements) are:
1. use `startStateNotifications` instead for the initial Bluetooth enabled checks. This will fire immediately with the current state when called, and will update as demanded when the adapter state changes.
2. call `isEnabled` twice when the application is first started.

```
  async isBluetoothEnabled(): Promise<boolean> {
    for (;;) {
      const result: { enabled: boolean; err?: any } = await ble.withPromises
        .isEnabled()
        .then(() => ({ enabled: true }))
        .catch((e) => ({ enabled: false, err: e }));

      if (!result.enabled && result.err == CBManagerState.Unknown && this.bleInitialising == true) {
        // Workaround for https://github.com/don/cordova-plugin-ble-central/issues/828
        // on iOS the adapter sometimes returns Unknown when queried immediately after initialising
        this.bleInitialising = false;
        await delay(100);
        continue;
      }

      this.bleInitialising = false;
      return result.enabled === true;
    }
  }
```

peitschie, 2022-02-23:
@tigohenryschultz with the demo app you've provided, I've been able to show that `ble.isEnabled` is working as I expect 

[image]

Is the true issue here the one raised by @nunoarruda ? I.e., the *first* call will almost always return disabled (because the adapter is warming up)?

tigohenryschultz, 2022-02-23:
Sounds like a possibility.

The bigger issue from the demo app on iOS was the startNotification not triggering a success after writing a command, but looks like the success for ble.write also never triggered for me on iOS

peitschie, 2022-02-24:
Based on the above, are you ok with closing this issue? The first-call-failing issue is already tracked in an existing issue, so not much value from keeping this open, in my opinion 🙂

tigohenryschultz, 2022-02-24:
> Based on the above, are you ok with closing this issue? The first-call-failing issue is already tracked in an existing issue, so not much value from keeping this open, in my opinion 🙂

Yeah that sounds fine by me.

closed 2022-02-24 by peitschie