don/cordova-plugin-ble-central #770: BLE.scan( ) doesn't work

teepad, 2020-06-08:
I'm trying use ble.scan, but I'm not getting any signal. Othes functions is working, like BLE.connect and BLE.enable

[image]

xdvarpunen, 2020-06-12:
I happened to have same issue. Calling ble instance methods in set interval somehow don't work.

Don't do ble instance method calls in the setInterval function.

xdvarpunen, 2020-06-12:
Use it more like this:

```javascript
constructor(ble){
    // 15 seconds is too little, put maybe 1min and stop from button?
    ble.scan([], 15, function(device) {
        console.log(JSON.stringify(device));
        ble.connect(device.id, function connectCallback(){
            // do stuff with connected device
        }, function disconnectCallback(){});
    }, failure);
}
```

don, 2020-11-08:
Don't use setInterval. This might be related to Android 10 permissions too. Fixed in v1.3.0.

closed 2020-11-08 by don