don/cordova-plugin-ble-central #772: iOS: Devices is not returned on Scan after connect to it.

JorgeMorata, 2020-06-09:
Hi!

In iOS, when you disconnect a Device, really are not disconnecting it and stay bounded. This makes that the device cannot be scanned more times.

More Details: 
In my app, I scan all the devices around me. I select one and I connect to it. When connect iOS ask you for bond the device (or you cannot connect to it). After that, i close de app (in this close I disconnect from device).

If I go to BT settings, the device appears still Connected so if I open the app another time, and do a Scan, the device not appears in the list so I cannot connect to it in my app.

Is there any solution in iOS please?

Thank you!

don, 2020-06-22:
Try [ble.connectedPeripheralsWithServices](https://github.com/don/cordova-plugin-ble-central#connectedperipheralswithservices)

don, 2020-11-08:
You should be able to disconnect. If your device is bonded, it may be causing the problem. On iOS you can get connected devices with [ble.connectedPeripheralWithServices](https://github.com/don/cordova-plugin-ble-central#connectedperipheralswithservices) or [ble.peripheralsWithIdentifiers](https://github.com/don/cordova-plugin-ble-central#peripheralswithidentifiers). If neither of these functions work, you might have an edge case that is better handled by writing a device specific plugin.

closed 2020-11-08 by don