don/cordova-plugin-ble-central #777: Works on Some Android Devices, Not on others

robertlair-lifequest, 2020-06-14:
I am testing my app on a Samsung Galaxy S8 and a LG Stylo 6.  The Galaxy S8 is running Android 9, and the Stylo is running Android 10.  Both popped up the permission for Location services, and both show location being enabled.  However, the ble.startScan() works on the Galaxy, but does nothing (no errors) on the Stylo.  Any thoughts on if this is an issue with the plugin, or if there is another issue.  

Thanks!

tiagoblackcode, 2020-06-17:
Hello, it might be because of Android 10 requiring background location permissions to scan BLE devices. Can you check #771 and test against [1.2.5-android-10](https://github.com/tiagoblackcode/cordova-plugin-ble-central/releases/tag/1.2.5-android-10) to see if it fixes the issue?

mcherubin, 2020-06-24:
Hello I'm using Cordova, I've try to use your suggestions  and adding 

    <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
    <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
    <uses-permission android:name="android.permission.ACCESS_BACKGROUND_LOCATION"/>

to my AndroidManifest.xml but startScan not working. I've try 1.2.5-android-10 to.

tiagoblackcode, 2020-06-25:
@mcherubin just to be sure, in the Permissions screen on the Android settings, check if there's a Background location permission for the app you're testing with. This is indicative that the manifest is correctly signaling the use of the background location permission and you can double-check you're not being bitten by cache issues.

don, 2020-11-08:
Fixed in v1.3.0

closed 2020-11-08 by don