don/cordova-plugin-ble-central #778: scan with a service UUID

FLATT-Dev, 2020-06-17:
When I start scan with an empty array of service ids the scan callback returns results
When enter the service UUID explicitly I get nothing
What's the proper syntax? The device has a 128 bit service UUID so I use

ble.startScanWithOptions(['00007A35-0000-1000-8000-00805F9B34FB'], {
				reportDuplicates : true
			}, function(data) {
				_this.scanCallback(data, null);
			}, function(err) {
				_this.scanCallback(null, err);
			});
Thanks

don, 2020-06-22:
That's the correct syntax. You're using the long form of a 16 bit UUID. Some Android devices are finicky and might want '7A35' as the UUID.

FLATT-Dev, 2020-06-22:
Thank you!
Andy

BetterAutomations, 2020-07-05:
FLATT-Dev did that work for you? I'm having the same problem.

FLATT-Dev, 2020-07-06:
Could be. No, passing the shorter form of the UUID did not work on Android. I ended up using an empty array .It does work on iOS

don, 2020-11-08:
This worked on a Pixel 1 with Android 10.
    
    const NORDIC_THINGY52_UUID = 'ef680100-9b35-4933-9b10-52ffa9740042';
    ble.startScanWithOptions(
         [ NORDIC_THINGY52_UUID ], 
         { reportDuplicates: true }, 
         peripheral => console.log(JSON.stringify(peripheral)),
         errorMessage => console.log(`ERROR ${errorMessage}`)
     )

I need to find a device advertising a 16-bit UUID. You might try to `toLowerCase()` on your UUID. It didn't make a difference for me but I real having a problem with UUID case in the past.

jrobeson, 2020-11-16:
I've seen some people say they needed to remove the hyphens for it to work on certain android devices. I can't verify that, but it might be worth a shot if you're still having problems.

GerardoPrototype, 2023-09-28:
Today I tried it with AA11 and aa11 did not get results, but when I sent [] as a parameter it had no problem. I tested it on an android 12.

I know it's an old problem but I hope my answer helps someone.

closed 2024-03-20 by peitschie