don/cordova-plugin-ble-central #780: call back is not called in ble.connect() in ios. but in android, it works.

vinothkumartyson, 2020-06-23:
ble.connect(myDeviceID, function(device) {
              alert("Successfully connected to device");
                console.log("Send Data method success call back called");
                var success = function() {
                    alert("Write Successfully");
                 };
                 var failure = function(err) {
                     alert("Failed writing data : "+JSON.stringify(err));
                 };
                  
                 var ID = '7';
                 var data = stringToBytes(ID);
                 var test = device.characteristics;
                 console.log(JSON.stringify(test[10]));
                 var chars = device.characteristics;
                 var service = chars[10].service;
                 var characteristics = chars[10].characteristic;
                 var deviceID = device.id;
                 console.log("Device ID : "+deviceID);
                 console.log("Serivce : "+service);
                 console.log("Device ID : "+characteristics);
                 console.log("DResult : "+JSON.stringify(device));
    
                 ble.writeWithoutResponse(deviceID,service,characteristics, data, success, failure);
             }, function(err) {
                console.log("Send Data method failed call back called");
                alert("Failed to connect"+JSON.stringify(err));
             });

don, 2020-06-23:
This is probably and app or device issue. The plugin works well with iOS. Maybe the connection is waiting for a timeout? Try resetting the peripheral. Turn Bluetooth on and off on the iOS device or reboot it. I recommend that you write a simple app the scans and connects to help debug this.

consolenaut, 2020-07-16:
I'm seeing the same issue too, building a basic scan & connect app to test the plugin out.
Scanning works great but when I call connect with the UUID of a device the callback is never fired. 
I've tried cycling the bluetooth & power-cycling the device with no luck, I've also set an interval to call isConnected every second & its returning as not connected. Is this maybe some iOS 13 weirdness where a key needs adding to the iOS info.plist to allow an app to connect to a bluetooth device or is "NSBluetoothAlwaysUsageDescription" sufficient?

Francois-Belanger, 2020-10-08:
Hi, I have the same issue with iOS.  I can see the connection is made, but the callback is never called :-(. Did you find a reason or a fix ?

Francois-Belanger, 2020-10-08:
Ok, this is a device bug, not iOS.  But still.. would have been fun for iOS to flag a mistake.  My device didn't adv any services...  Thanks for the great work !