don/cordova-plugin-ble-central #781: Report duplicates does not work when setting a service uuid on Android.

jrobeson, 2020-06-24:
```js
ble.startScanWithOptions([], { reportDuplicates: true}, (peripheral => console.log(JSON.stringify(peripheral)) }
```
works, while 
```js
ble.startScanWithOptions([/* relevant uuid */], { reportDuplicates: true}, (peripheral => console.log(JSON.stringify(peripheral)) }
```
does not.  I looked through the js code and the java code and nothing looks like it should be causing it.

don, 2020-11-08:
@jrobeson can you verify the peripheral is advertising the service you specified as your relevant UUID? Make sure it's a service UUID, not a device UUID.

        const NORDIC_THINGY52_UUID = 'ef680100-9b35-4933-9b10-52ffa9740042';
        ble.startScanWithOptions(
             [ NORDIC_THINGY52_UUID ], 
             { reportDuplicates: true }, 
             peripheral => console.log(JSON.stringify(peripheral)),
             errorMessage => console.log(`ERROR ${errorMessage}`)
         )

don, 2020-11-08:
This looks like a duplicate of #778

jrobeson, 2020-11-08:
yeah, seems so.

closed 2020-11-08 by jrobeson