don/cordova-plugin-ble-central #789: Starting to receive notifications fails

Juarrow, 2020-07-16:
(I'm using Android 9)

Every time i try to subscribe to notifications i get the error

> Set notification failed for 000036f6-0000-1000-8000-00805f9b34fb

But when i register for notifications from an native app on the same characteristic, it works just fine.

SSDevAdmin, 2020-10-22:
I have used the startNotification successfully with devices implementing BLE 4.1 and 4.2 on Android 8.1 platform. When I use the same code in Android platform 9 or above, the BLE 4.1 device does not return anything (success or error) - the BLE 4.2 device continues to work fine (it is not the location permission etc. which are set). 

Both the BLE 4.1 and 4.2 device scan, connect, disconnect, read, write works. This issue is only for the startNotification. Not sure what might be different.

nolife08021, 2021-01-13:
@SSDevAdmin  @Juarrow  I facing the same issue, did you guys fixed it?

SSDevAdmin, 2021-01-13:
I temporarily did a timer based loop read of the characteristics (they were read and notify)... not the desired solution, but was a workaround. The original notification is still persisting.

nolife08021, 2021-01-13:
oh, okay i will try it if still haven't found any solution. thanks

lon2347, 2021-01-31:
By using Wireshark, I found the value of notifiction is still "0" not "1"after running ble.startnotification() on my andriod phone, that means notify is disable and any data can not be received from periperal, seems function ble.startnotification did not set CCCD descriptor.

peitschie, 2022-06-29:
Is this issue still occurring for anyone?

peitschie, 2024-01-29:
Closing due to inactivity

closed 2024-01-29 by peitschie