don/cordova-plugin-ble-central #791: write data error callback 128

akashkhairnar01, 2020-07-20:
i am getting the write data error callback 128. how to solve this error. 
my service = '6e400001-b5a3-f393-e0a9-e50e24dcca9e' and
characteristic = '6e400002-b5a3-f393-e0a9-e50e24dcca9e'.

akashkhairnar01, 2020-07-21:
hi @don please help me to solve this issue, why i am getting this issue . i have searched on internet it said that descriptor is not set but in plugin the descriptor is already set. what can i do to solve this error?

tellybrown, 2020-11-06:
What you will want to do is look at the connect response and look at the characteristics.  They have permissions array that comes along with each characteristic.  You might make sure it can write.  I have a device and there is a TX and a RX characteristic and another for notifications.  You have to use each one for its given purpose.

don, 2020-11-08:
Check the permissions like @tellybrown mentioned. An error of 128 might be 0x80 GATT_NO_RESOURCES which I think is a problem in the peripheral.

closed 2020-11-08 by don