don/cordova-plugin-ble-central #800: Sending longer commands doesnt work on Samsung S7

LocutusOfPenguin, 2020-08-18:
The ble hardware i want to communicate with has small commands and a long 167 bytes one. Each is with party bytes at end to check the commands.

My App works fine, if i send <10bytes commands (incl. s7) and 167bytes commands to lots of mobiles.
The sending is split inside my app with 20bytes chunks. I use ble.send for this.

But when i try to send such long (167bytes) command to a samsung S7 (Android 8), the ble hw on the other side behaves like not received this command (it will ignore it, if the party bytes of command not ok, or just nonsense).

I also tried to enter wait cylces between these 20bytes chunks, changing the 20 to 10, 15 bytes...nothing works on samsung s7 ONLY. Also i tried other android 8 mobiles..no problem.

I dont know, what else i can try. Do you have an idea what is the cause of this?

peitschie, 2022-02-15:
There is no built-in chunking to the Bluetooth standard, so it's possible that the BLE hardware has set a maximum transmission unit (which dictates the max packet size) lower than 167 bytes.

It might be worth using nrf Connect or a similar application to see if you can determine what the default MTU being negotiated between the phone and the BLE hardware is. This number determines the largest packet that can be sent, and may be the reason a 167 byte payload is being ignored.

I'll close this out for now as there's no obvious shortcoming in the plugin here. However, feel free to continue the discussion or re-open this if it's still an issue.

closed 2022-02-15 by peitschie

LocutusOfPenguin, 2022-02-15:
No, mainly this is only possible on android. But I need android & iOS.
This issue is very old, and as I said the ONLY device not working is S7 and only with a long command.
Every other device (and I tried a lot) worked. Therefore I found it strange.

My solution was to not do another command anymore. Well, right now I can't remember which :-) - as I said, old issue.
Couldn't wait, and I desperately tried a lot of combinations, until it worked. But I still can't explain why this happening.
Practical answer:
But for me, this is working now...incl. the S7, so I continue with something else. Thanks anyway.

peitschie, 2022-02-15:
No worries at all! Thanks for the update @LocutusOfPenguin ... always interesting to see where things end up, even with old issues. If you hit any other issues, don't hesitate to raise a new issue to discuss.