don/cordova-plugin-ble-central #803: not scanning devices on android 10 Api level 29

bluemardia, 2020-08-28:
When I set  targetSdkVersion to 29 in build.gradle, the ble startscan method no longer returns a success or failure.

smartyw, 2020-08-29:
I have this problem too.

Eskilax, 2020-08-31:
I had the same problem.

I have been able to solve it by forcing a gps request before the scan.

Installing this plugin: _**cordova plugin add cordova-plugin-geolocation**_

And using this code before scan: _**navigator.geolocation.getCurrentPosition (function () {}, function () {});**_

That worked for me.

bluemardia, 2020-08-31:
> I had the same problem.
>
> I have been able to solve it by forcing a gps request before the scan.
>
> Installing this plugin: _**cordova plugin add cordova-plugin-geolocation**_
>
> And using this code [...]

Amazing, this seems to have worked. Will sleep much better tonight. THANK YOU SO MUCH!

dslima90, 2020-09-01:
I also have this problem. The GPS trick seems to work, but when are we going to have a fix for this?

vicatcu, 2020-09-02:
Unfortunately, this workaround causes the app to require a GPS in the mobile device, which drops support for about 1451 devices according to the Google Play console for my app. I guess on the upside it will actually work for the rest...

dslima90, 2020-09-03:
> Unfortunately, this workaround causes the app to require a GPS in the mobile device, which drops support for about 1451 devices according to the Google Play console for my app. I guess on the upside it will actually work for the rest...

There is already a pull request that solves the issue without having to use the GPS trick.

mpacary, 2020-09-03:
I first tried [this fork](https://github.com/tkem/cordova-plugin-ble-central), which has same changes as the [pull request](https://github.com/don/cordova-plugin-ble-central/pull/798), but `ble.scan()` and `ble.startScan()` still did not detect BLE peripherals (no call to success or error callbacks).

Only the "GPS workaround" has been effective on my Android 10 phone, using Cordova with defaultTargetSdkVersion=29,  cordova-android 9.0.0, Gradle 6.5

dslima90, 2020-09-04:
> I first tried [this fork](https://github.com/tkem/cordova-plugin-ble-central), which has same changes as the [pull request](https://github.com/don/cordova-plugin-ble-central/pull/798), but [...]

@mpacary I did the same changes (Changed COARSE_LOCATION do FINE_LOCATION permission) myself in a [fork](https://github.com/dslima90/cordova-plugin-ble-central) of this repo and it seems to work in my Android 10 Motorola One phone with android-cordova 9.0.0. Maybe you missed something?

markamccorkle, 2020-09-04:
> Unfortunately, this workaround causes the app to require a GPS in the mobile device, which drops support for about 1451 devices according to the Google Play console for my app. I guess on the upside it will actually work for the rest...

`For historic reasons, this plugin requires GPS Hardware on Android devices, so your app will not be able to run on devices without. If you want to use this plugin in your app, but you do not require actual GPS Hardware on the device, install the plugin using the variable GPS_REQUIRED set to false:`

cordova plugin add cordova-plugin-geolocation --variable GPS_REQUIRED="false"

hicksy, 2020-09-08:
> I had the same problem.
>
> I have been able to solve it by forcing a gps request before the scan.
>
> Installing this plugin: _**cordova plugin add cordova-plugin-geolocation**_
>
> And using this code [...]

Thanks so much! I've been pulling my hair out on this one for months. 

FYI for anyone using `@ionic-native/ble` in a Capacitor project, which wraps `cordova-plugin-ble-central` (https://ionicframework.com/docs/native/ble)... you can achieve the same fix like so:

Use the Geolocation Capacitor plugin e.g
```
const { Geolocation } = Plugins;
```

The before you begin a bluetooth scan:
```
const coordinates = await Geolocation.getCurrentPosition();
```

rajeshpandalss, 2020-09-08:
@hicksy I tried the solution you proposed for capacitor project. Doesn't work for me. I am targeting 29 on Android 10.

dslima90, 2020-09-23:
Using this fork it works ok on android 10, but stops working on android 9... No official solution for this?

chris-guidry, 2020-10-05:
@tkem's solution worked in my fork of the code, and addresses the root of the issue rather than installing a geo location plugin as a work around. I found that I needed to update my AndroidManifest.xml file as well because it referenced the coarse location setting.

tkem, 2020-10-05:
@rajeshpandalss, @chris-guidry: Well, this isn't exactly "my" solution, but based on PR #798. So @subratpalhar92 is the one who deserves credit for this!

ASHBAW, 2020-10-05:
Just to make sure I have the right fork. Is this the correct one:

"cordova-plugin-ble-central": "git+https://github.com/rajeshpandalss/cordova-plugin-ble-central.git"

angel-garcia-lopez, 2020-10-16:
> > I first tried [this fork](https://github.com/tkem/cordova-plugin-ble-central), which has same changes as the [pull request](https://github.com/don/cordova-plugin-ble-central/pull/798), but `ble.scan()` and `ble.startScan()` still did not detect BLE peripherals (no call to success or error callbacks).
> > Only the "GPS workaround" has been effective on my Android 10 phone, using Cordova with defaultTargetSdkVersion=29, cordova-android 9.0.0, Gradle 6.5
> 
> @mpacary I did the same changes (Changed COARSE_LOCATION do FINE_LOCATION permission) myself in a [fork](https://github.com/dslima90/cordova-plugin-ble-central) of this repo and it seems to work in my Android 10 Motorola One phone with cordova-android 9.0.0. Maybe you missed something?

For me this is clearly the correct answer. As stated in the docs (https://developer.android.com/guide/topics/connectivity/bluetooth#Permissions) you now need to provide FINE_LOCATION to get it working. I changed it locally and got it working now. I hope they fix this officially.

rajeshpanda, 2020-10-16:
> Just to make sure I have the right fork. Is this the correct one:
> 
> "cordova-plugin-ble-central": "git+https://github.com/rajeshpandalss/cordova-plugin-ble-central.git"

I created a fork of it to avoid issues down the line. I suggest you do the same. @ASHBAW

don, 2020-11-08:
Fixed in v1.3.0

closed 2020-11-08 by don