don/cordova-plugin-ble-central #805: startNotification response empty [{},2] @android8.0.0

butlet, 2020-09-04:
hi,everyone. i am building an android app using ionic5 framework in which cordova-plugin-ble-central plugin is used. The BLE peripheral has a characteristic which pushes notifications to phone.
i enable the notification and register the callback by calling startNotification(),when the ble sends message, the phone just received [{},1],or [{},2],[{},3].... why no data,just has received times

`  start_notification(){
    this.ble.startNotification(this.connected_device.id, SERVICE_ID_READ,CHARACTERISTICS_ID_READ).subscribe(
      msg => {
        this.setStatus("notification succ:" + JSON.stringify(msg));
        // let data = new Uint8Array(msg);
        // this.setStatus("notification succ:" + data[0]);
      },
      err => {
        this.setStatus("notification err:" + JSON.stringify(err));
      } 
    )
  }`

saayalac, 2020-09-08:
I am facing the same problem. I had to go to an older version to make it work. Had the problem with android 6, 9 and 10

markamccorkle, 2020-09-20:
Have you tried reading the first index in the response to an array buffer? This was changed a little while back and nobody updated the docs. 

`const data = new Uint8Array(buffer[0]);`

Bozonych, 2020-10-21:
> Have you tried reading the first index in the response to an array buffer? This was changed a little while back and nobody updated the docs.
> 
> `const data = new Uint8Array(buffer[0]);`

yep, i will changed my code:
from:              ```const data16 = new Uint16Array(buffer);```
to:              ```const data16 = new Uint16Array(buffer[0]);```
and all works fine ^-^

closed 2020-10-22 by butlet