don/cordova-plugin-ble-central #808: Unbond Method

scrabionau77, 2020-09-20:
Hi everyone.
Is there (or is it possible to have) a method that allows the deletion of bonding information?
(this functionality is for example present in "NRF Connect" and can be very useful in app development)

scrabionau77, 2020-09-29:
I am trying to implement the "unbond" function. Do you think it can work like this? I would put this code in the file BLECentralPlugin.java

`public void unbond(CallbackContext callbackContext, String macAddress){
        if (!peripherals.containsKey(macAddress) && BLECentralPlugin.this.bluetoothAdapter.checkBluetoothAddress(macAddress)) {
            callbackContext.error("Peripheral " + macAddress + " not already bonded.");
            return;
        }

        Peripheral peripheral = peripherals.get(macAddress);
        if (peripheral != null) {
            try{
                Method m = peripheral.getClass().getMethod("removeBond", (Class[]) null);
                m.invoke(peripheral, (Object[]) null);
                callbackContext.success();
            } catch (Exception e){
                callbackContext.error("Not unbonded");
            }
            
        } else {
            callbackContext.error("Peripheral " + macAddress + " not found.");
        }
    }`

chris-guidry, 2020-10-05:
I did something similar in my fork of the code. I added [the function unpairDevice to Peripheral.java](https://github.com/chris-guidry/cordova-plugin-ble-central/blob/master/src/android/Peripheral.java#L148) and then call that function before connecting. Certain LG and Motorola phones don't handle bonding well so they will connect once, but not again until the bonding is deleted.

scrabionau77, 2020-10-08:
After some hours, my worked method in BLECentralPlugin.java is:

`public void unbond(CallbackContext callbackContext, String macAddress){
        LOG.w(TAG, "Estraggo lista bracciali bondati. Cerco: " + macAddress);
        Set<BluetoothDevice> bondedDevices =  bluetoothAdapter.getBondedDevices();
        List<BluetoothDevice> ListP = new ArrayList<BluetoothDevice>();
        for (BluetoothDevice device : bondedDevices) {
            device.getBondState();
            int type = device.getType();
            // just low energy devices (filters out classic and unknown devices)
            if ((type == DEVICE_TYPE_LE || type == DEVICE_TYPE_DUAL)) {
                if(device.getAddress().trim().equals(macAddress.trim())){
                    ListP.add(device);
                }
            }
        }
        if(ListP.size() == 0){ // no device found
            LOG.w(TAG, "Empty list");
            callbackContext.error("Peripheral " + macAddress + " not already bonded.");
            return;
        }
        BluetoothDevice device = ListP.get(0);
        if (device != null) {
            boolean result = false;
            try {
                java.lang.reflect.Method mi = device.getClass().getMethod("removeBond");
                Boolean returnValue = (Boolean) mi.invoke(device);
                result = returnValue.booleanValue();
            } catch (Exception e) {
                LOG.e(TAG, "UNBONDING ERROR: " + e.getMessage());
                callbackContext.error("Not unbonded");
            }
            if(result){
                callbackContext.success();
            } else {
                LOG.e(TAG, "UNBONDING ERROR");
                callbackContext.error("Not unbonded");
            }
        } else {
            callbackContext.error("Peripheral " + macAddress + " not already bonded.");
        }
    }`

peitschie, 2022-02-07:
Thanks for documenting your workaround here @scrabionau77 !

It's worth noting that this approach may cause problems in the future. Google have been slowly cracking down on use of private APIs such as this. I'd suggest this approach is used with some care, as there is a risk of an app being delisted.

closed 2022-02-07 by peitschie