don/cordova-plugin-ble-central #811: Randomly discconects

YaariBoaz, 2020-10-15:
Hey,
I'm using the plugin with ionic 4 and angular 4.

I'm having issues with random disconnections.

All fo the BLE logic is held in a service, this is my connect function:

  ```
connect(deviceId) {
        if (this.gateways.indexOf(deviceId) > -1) {
            this.isGateway = true;
            this.initService.isGateway = true;
        }
        this.currentTargetId = deviceId;
        this.subscription = this.ble.connect(deviceId).subscribe(
            (peripheral) => {
                this.isConnectedFlag = false;
                this.notifyTargetConnected.next(true);
                this.onConnected(peripheral);
            },
            peripheral => {
                console.log('Disconnected', 'The peripheral unexpectedly disconnected');
                this.activatRecconectProcess();
            }, () => {
            }
        );
    }
```

I tested the same Android device with the same BLE device with C# code and it doesn't disconnect.
So that eliminates the peripherals issue.

I'm guessing my error is somewhere in the code.

Thanks in advance.

tellybrown, 2020-11-06:
I have seen disconnects when I didn't handle both success and failure events and an error dropped the connection.  What you might do instead is use the autoconnect method to connect to the id vs trying to manually control it.  Thats what I did and it worked much better IMO.

peitschie, 2023-03-16:
Closing due to inactivity and a lack of further detail.

@tellybrown's tip about using autoConnect is reasonable, just worth being aware that the auto reconnection loop doesn't survive some events such as Bluetooth disable.

If there are still ongoing issues here, feel free to re-open the issue with some more details and logs!

closed 2023-03-16 by peitschie