don/cordova-plugin-ble-central #817: IONIC - Prior issues with getting characteristics and services

tellybrown, 2020-10-29:
So I have done some digging around my issue and it seems others had issues as well.

I can't say that all of these are the same issue but they all are trying to find the characteristics when scanning or connecting.
https://github.com/don/cordova-plugin-ble-central/issues/501
https://github.com/don/cordova-plugin-ble-central/issues/461
https://github.com/don/cordova-plugin-ble-central/issues/618
https://github.com/don/cordova-plugin-ble-central/issues/657
https://github.com/don/cordova-plugin-ble-central/issues/806
https://github.com/don/cordova-plugin-ble-central/issues/774

1. This has been answered before.  You do not receive services and characteristics from a scan.  **You must connect**.
2. If you are using IONIC like I am... and I am on the latest IONIC 5 + Capacitor, it defaults to Android API 21.  Which is less than the API version in the plugin to determine to use BLE or not.  [Here](https://github.com/don/cordova-plugin-ble-central/blob/35108d43e66ccd576884daef472e8f05ffa93729/src/android/Peripheral.java#L98)
Go into the config.xml or variables.gradle or androidmanifest.xml to alter based upon your version of IONIC.  I simply up'd mine to 23 and it works now.

peitschie, 2021-11-20:
Thanks for the summary work @tellybrown !

Your understanding is correct in that the services and characteristics are not contained within the advertising data received during a scan. You must indeed connect to get the full list of these.

I'll leave this open for now, as there might be some benefit to clarifying with Ionic what their expectations are in terms of default API versions and best support.

That interesting to hear that the default Ionic setup is not terribly compatible with this plugin. These days, I'd highly suggest aiming at API 26+ for best compatibility.

peitschie, 2023-03-16:
Closing this out, as the README is now a bit clearer about the minimum supported SDK versions for Android.

closed 2023-03-16 by peitschie