don/cordova-plugin-ble-central #818: I have problems connecting an android device to another device

DesCreategaDavid, 2020-10-30:
Que tal buen dia, estoy haciendo una aplicacion para poder conectar y emparejar dispositivos android a cualquier otro dispositivo, por medio de bluetooth, la verdad me interesa que se conecte a un dispositivo android y a una pc con windows 10. Ya desarrolle mi codigo sin problemas, ya pude escanear y obtengo los dispositivos que estan cercas de mi o los que se encuentran visibles, pero, solo obtengo los dispositivos de pc windows 10, tengo otro dispositivo con android y a la hora de escanear en mi lista no aparece ese dispositivo.

Cuando me quiero conectar a un dispositivo, no sale ninguna respuesta.
Espero y me puedan ayudar.
Este  es el complemento que utilizo.
import { BLE } from '@ionic-native/ble/ngx';

import { BLE } from '@ionic-native/ble/ngx';
import { Component, NgZone } from '@angular/core';
import { NavController, ToastController } from '@ionic/angular';
import { NavigationExtras, Router } from '@angular/router';

@Component({
  selector: 'app-home',
  templateUrl: 'home.page.html',
  styleUrls: ['home.page.scss'],
})
export class HomePage {
  devices: any[] = [];
  mensaje: string;
  deviceMode = true;
  peripheral: any = {};
 
  constructor(
    public router: Router,
    private toastCtrl: ToastController,
    private ble: BLE,
    private ngZone: NgZone
  ) {}

  ionViewDidEnter() {
    console.log('ionViewDidEnter');
    this.scan();
  }

  scan() {
    this.setStatus('Escaneando dispositivos');
    this.devices = []; // clear list

    this.ble.scan([], 60).subscribe(
      device => this.onDeviceDiscovered(device),
      error => this.scanError(error)
    );

    setTimeout(this.setStatus.bind(this), 5000, 'Escaneo completo');
  }

  onDeviceDiscovered(device) {
    console.log('Discovered ' + JSON.stringify(device, null, 2));
    this.ngZone.run(() => {
      this.devices.push(device);
    });
  }

  // If location permission is denied, you'll end up here
  async scanError(error) {
    this.setStatus('Error ' + error);
    const toast = await this.toastCtrl.create({
      message: 'Error al escanear los dispositivos',
      position: 'middle',
      duration: 5000
    });
    toast.present();
  }

  setStatus(message) {
    console.log(message);
    this.ngZone.run(() => {
      this.mensaje = message;
    });
  }

  deviceSelected(device: any) {
  
    this.ble.disconnect(device.id).then(
      () => console.log('Desconectado ' + JSON.stringify(this.peripheral)),
      () => console.log('Error en la conexión ' + JSON.stringify(this.peripheral)));
  }

  onConnected(peripheral) {
    this.ngZone.run(() => {
      this.setStatus('');
        this.peripheral = peripheral;
        
    });
  }

 async onDeviceDisconnected(peripheral) {
    const toast = await this.toastCtrl.create({
      message: 'Esta desconectado',
      duration: 3000,
      position: 'middle'
    });
    toast.present();
  }

}

tellybrown, 2020-10-30:
Probably your location has to be enabled for discovery.  Here is my method for requesting perms and turning on ble.

    async askPermission(loc: boolean) {
        if (Capacitor.platform == 'android') {
            var btPerm = await this.androidPermissions.checkPermission(this.androidPermissions.PERMISSION.BLUETOOTH).catch(async (err) => {
                await this.androidPermissions.requestPermission(this.androidPermissions.PERMISSION.BLUETOOTH);
            });

            if (loc) {
                var fineLocPerm = await this.androidPermissions.checkPermission(this.androidPermissions.PERMISSION.ACCESS_FINE_LOCATION).catch(async (err) => {
                    await this.androidPermissions.requestPermission(this.androidPermissions.PERMISSION.ACCESS_FINE_LOCATION);
                });

                var canRequest = this.locationAccuracy.canRequest();
                if (canRequest) {
                    await this.locationAccuracy.request(this.locationAccuracy.REQUEST_PRIORITY_HIGH_ACCURACY).catch(err => console.log(err));
                }
            }

            await this.ble.enable();
        }
    }

DesCreategaDavid, 2020-10-30:
@tellybrown good day thanks for supporting me, add this part of code that you provided me, apparently everything is in order but I still get that it is offline.
It should be noted that the devices that I want to connect are enabled and that they can connect.
But through the code presented above, I don't know what else I am missing.
 
Below I provide some images.
[image]

tellybrown, 2020-10-30:
Are those other devices BLE? Or Serial?  I don't think this library does serial which is what I have done in the past with Arduino based BT devices.  Serial is MUCH simpler to get working but I am pretty sure requires a different library.

DesCreategaDavid, 2020-10-30:
@tellybrown. So also I already tried it by serial connection, using the plugins, but I was unsuccessful, if I use that plugin it also shows me the list of those that are close and ready to pair, but when I try to connect with the device I get an error.
Also what I have seen is that these plugins that Ionic handles are very new versions, and in the forums I do not find much information about it.

I already implemented these plugins in other demo projects, without success, I don't know if someone has a demo that can help me.

https://ionicframework.com/docs/native/bluetooth-le
https://ionicframework.com/docs/native/bluetooth-serial

tellybrown, 2020-10-30:
Only way i can help is to know what device you are trying to connect to.

DesCreategaDavid, 2020-10-30:
ok it's a laptop Dell, with windows 10, and a Samsung J5 cell phone.

tellybrown, 2020-10-31:
check your laptop to see what version it has. 
This says certain versions only have BT 3.0 and not BTLE
https://www.dell.com/community/Networking-Internet-Bluetooth/Bluetooth-4-0-LE/td-p/5175161

DesCreategaDavid, 2020-11-02:
@tellybrown Thanks for helping me, the firmware version is
HCI 9.0 / LMP 9.602

tellybrown, 2020-11-02:
I am not sure what the issue is then.  You might try to see if windows can detect the device through its system bluetooth settings.

don, 2020-11-07:
The computer or phone that you want your app to connect to needs to be acting as a BLE peripheral and advertising BLE service. You can make another Android phone a peripheral by writing a Cordova app with [cordova-plugin-ble-peripheral](http://github.com/don/cordova-plugin-ble-peripheral). For testing you can use an app like [nRF Connect](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp). nRF Connect can be a central device and scan for peripherals. Or act as a peripheral and be discovered by central devices like your app. Note you need 2 phones for this, the phone can't discover itself over BLE. It looks like there's [nRF Connect for Desktop](https://www.nordicsemi.com/Software-and-tools/Development-Tools/nRF-Connect-for-desktop). I haven't used this, but it might help you.

peitschie, 2022-02-15:
I'm closing this out for now as there's no obvious shortcoming in the plugin to be addressed.

However, feel free to continue the conversation here or re-open if I've missed a use-case here.

closed 2022-02-15 by peitschie