don/cordova-plugin-ble-central #819: can't compile 1.3.0 (in framework7)

LocutusOfPenguin, 2020-11-08:
Works on 1.2.5 but not 1.3.0 I get the following errors:

/Users/locutusofpenguin/VueCl7/cordova/platforms/android/app/src/main/java/com/megster/cordova/ble/central/BLECentralPlugin.java:661: error: cannot find symbol
        if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.Q) {
                                                        ^
  symbol:   variable Q
  location: class VERSION_CODES
/Users/locutusofpenguin/VueCl7/cordova/platforms/android/app/src/main/java/com/megster/cordova/ble/central/BLECentralPlugin.java:669: error: cannot find symbol
                        Manifest.permission.ACCESS_BACKGROUND_LOCATION
                                           ^
  symbol:   variable ACCESS_BACKGROUND_LOCATION
  location: class permission
Note: /Users/locutusofpenguin/VueCl7/cordova/platforms/android/app/src/main/java/com/megster/cordova/ble/central/BLECentralPlugin.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
2 errors

FAILURE: Build failed with an exception.

don, 2020-11-08:
I expect I'm using a newer version of Android SDK than you or it's an API level thing or compile version or something. 

What versions of Cordova and Android SDK are you using? I'll see if I can duplicate and maybe add some conditional logic.

LocutusOfPenguin, 2020-11-08:
i use framework7(.io) and i had cordova 9.0.0 which i now updated to 10.0.0
Also in the android studio i downloaded the sdk platform "android 10.0 (Q)" but this all has no effect until now.

don, 2020-11-08:
I don't think this is related to framework7. Android 10 and API 29 should fix this, but I expect your project is using cordova-android@8.0.0. You can list the platforms to check.

```
C:\Users\don\foo-cordova>cordova platform ls
Installed platforms:
  android 9.0.0
Available platforms:
  browser ^6.0.0
  electron ^1.0.0
  windows ^7.0.0
```

cordova-android-9 was released in June. I'm not sure how many people are still running cordova-android-8. I'll make a branch and see if I can support older versions. It would be good if you could test this for me before upgrading your project. I'll post a link here.

LocutusOfPenguin, 2020-11-08:
That solved the problem! Before it showed 8.1.0 and I removed & installed android platform again..now 9.0.0 and it compiles (haven't checked the resulting file) but both ways working just deprecated warnings.

don, 2020-11-08:
I think these changes will allow the plugin to with older versions

    cd path/to/project
    cordova plugin rm cordova-plugin-ble-central
    cordova plugin add https://github.com/don/cordova-plugin-ble-central#819-less-than-Q

don, 2020-11-08:
Confirmed changes work with cordova-android@9.0.0 and cordova-android@8.1.0. This should be in 1.3.1

closed 2020-11-08 by don

LocutusOfPenguin, 2020-11-08:
thanks, don!

PR #820 (2020-11-08): compile with android API less than 29
Fixes #819