don/cordova-plugin-ble-central #823: Feature Request: isConnectable Property for Android

IngoKoch, 2020-11-23:
Advertising Information on iOS offers a flag isConnectable (kCBAdvDataIsConnectable).
Would it be possible to issue such a flag on Android as well? As far as I know it is available since Android 8.0.
Source: https://stackoverflow.com/questions/28467590/detecting-whether-a-ble-device-is-connectable-on-android

atuljain2995, 2021-03-25:
I am stuck in determining whether BLE device is connectable or not, by analyzing advertisement data. Please help how to find the PUD from advertisement data?

peitschie, 2021-11-20:
This is related to #192.

It's possible to parse this manually out of the advertising payload directly, but it is a bit tricky.
I agree there may be some benefit to exposing more of the properties from https://developer.android.com/reference/android/bluetooth/le/ScanResult.html#isConnectable().

Patches to do this would be welcome (especially if they update documentation and the iOS implementation also for consistency!)

closed 2023-11-03 by peitschie