don/cordova-plugin-ble-central #824: Writing does not work on IOS

jensuenk, 2020-12-01:
Hello,

I'm creating an ionic app which can connect to a Bluno microcontroller (bluetooth powered Arduino). I ran into a problem when writing data from the app on ios. When I run the write function I get no callbacks at all.
The weird thing is that it works perfectly on android. There I get all data in my Arduino serial monitor and a success callback.

I am running all latest versions, IOS 14.3, and the latest firmware of the Bluno

Any ideas what is going on here?

`this.ble.write(this.device.id, 'dfb0', 'dfb1', data)
      .then(function (result) {
        console.log(result)
      })
      .catch(function (error) {
        console.log(error)
      });`

Json device data:
`[ng] [console.log]: {
[ng]   "name": "Bluno",
[ng]   "id": "50:65:83:6F:8F:FE",
[ng]   "advertising": {},
[ng]   "rssi": -43,
[ng]   "services": [
[ng]     "1800",
[ng]     "1801",
[ng]     "180a",
[ng]     "dfb0"
[ng]   ],
[ng]   "characteristics": [
[ng]     {
[ng]       "service": "1800",
[ng]       "characteristic": "2a00",
[ng]       "properties": [
[ng]         "Read"
[ng]       ]
[ng]     },
[ng]     {
[ng]       "service": "1800",
[ng]       "characteristic": "2a01",
[ng]       "properties": [
[ng]         "Read"
[ng]       ]
[ng]     },
[ng]     {
[ng]       "service": "1800",
[ng]       "characteristic": "2a02",
[ng]       "properties": [
[ng]         "Read",
[ng]         "Write"
[ng]       ]
[ng]     },
[ng]     {
[ng]       "service": "1800",
[ng]       "characteristic": "2a03",
[ng]       "properties": [
[ng]         "Write"
[ng]       ]
[ng]     },
[ng]     {
[ng]       "service": "1800",
[ng]       "characteristic": "2a04",
[ng]       "properties": [
[ng]         "Read"
[ng]       ]
[ng]     },
[ng]     {
[ng]       "service": "1801",
[ng]       "characteristic": "2a05",
[ng]       "properties": [
[ng]         "Indicate"
[ng]       ],
[ng]       "descriptors": [
[ng]         {
[ng]           "uuid": "2902"
[ng]         }
[ng]       ]
[ng]     },
[ng]     {
[ng]       "service": "180a",
[ng]       "characteristic": "2a23",
[ng]       "properties": [
[ng]         "Read"
[ng]       ]
[ng]     },
[ng]     {
[ng]       "service": "180a",
[ng]       "characteristic": "2a24",
[ng]       "properties": [
[ng]         "Read"
[ng]       ]
[ng]     },
[ng]     {
[ng]       "service": "180a",
[ng]       "characteristic": "2a25",
[ng]       "properties": [
[ng]         "Read"
[ng]       ]
[ng]     },
[ng]     {
[ng]       "service": "180a",
[ng]       "characteristic": "2a26",
[ng]       "properties": [
[ng]         "Read"
[ng]       ]
[ng]     },
[ng]     {
[ng]       "service": "180a",
[ng]       "characteristic": "2a27",
[ng]       "properties": [
[ng]         "Read"
[ng]       ]
[ng]     },
[ng]     {
[ng]       "service": "180a",
[ng]       "characteristic": "2a28",
[ng]       "properties": [
[ng]         "Read"
[ng]       ]
[ng]     },
[ng]     {
[ng]       "service": "180a",
[ng]       "characteristic": "2a29",
[ng]       "properties": [
[ng]         "Read"
[ng]       ]
[ng]     },
[ng]     {
[ng]       "service": "180a",
[ng]       "characteristic": "2a2a",
[ng]       "properties": [
[ng]         "Read"
[ng]       ]
[ng]     },
[ng]     {
[ng]       "service": "180a",
[ng]       "characteristic": "2a50",
[ng]       "properties": [
[ng]         "Read"
[ng]       ]
[ng]     },
[ng]     {
[ng]       "service": "dfb0",
[ng]       "characteristic": "dfb1",
[ng]       "properties": [
[ng]         "Read",
[ng]         "WriteWithoutResponse",
[ng]         "Write",
[ng]         "Notify"
[ng]       ],
[ng]       "descriptors": [
[ng]         {
[ng]           "uuid": "2901"
[ng]         }
[ng]       ]
[ng]     },
[ng]     {
[ng]       "service": "dfb0",
[ng]       "characteristic": "dfb2",
[ng]       "properties": [
[ng]         "Read",
[ng]         "WriteWithoutResponse",
[ng]         "Write",
[ng]         "Notify"
[ng]       ],
[ng]       "descriptors": [
[ng]         {
[ng]           "uuid": "2901"
[ng]         }
[ng]       ]
[ng]     }
[ng]   ]
[ng] }`

rogerfachini, 2020-12-01:
What's the length in bytes of the data you're trying to write? I encountered similar symptoms; writes would work on Android but not on iOS with large chunks of data (>99 bytes in my particular case). Smaller writes would work on both systems.

jensuenk, 2020-12-01:
Just a single integer converted to bytes.

closed 2020-12-14 by jensuenk

Path-A, 2021-03-22:
Were you able to fix this @jensuenk?