don/cordova-plugin-ble-central #825: Peripheral does not disconnect on app close or bluetooth disable.

Domvel, 2020-12-01:
I have a weird problem with the BLE connection. It doesn't disconnect when I quit the app or when Bluetooth is turned off. Yes, not even when restarting the tablet. But it works with other bluetooth apps. So I guess it's a problem with this plugin or Cordova.

I don't know exactly how low-level bluetooth works. How does a disconnect work? Specifically when exiting the app that made the connection or how it works when bluetooth is off. 
If I see that correctly, an event is sent during these actions (exit app / switch off BT). But not sure? I just wonder, why does it work with other apps like "nRF Connect"? Only when connecting with the Ionic / Cordova app with this plugin causes this problem.

Any guesses / ideas / experiences? 

I'll do more test about this issue. I create a new app and test the connection. 
My current solution is a custom heartbeat sent every seconds. And a timeout causes a disconnect.
But I don't want such special implementation. ...

@don I hope I don't bother you. But I would be glad if you could say something about it.

Versions:
`cordova-plugin-ble-central`: `1.3.1`
`@ionic-native/ble`: `5.30.0`
`@ionic/angular`: `5.3.2`
`cordova-android`: `9.0.0`
`cordova-plugin-androidx-adapter`: `1.1.1`
`Android`: `7`

config.xml (snippet):

```
<preference name="ScrollEnabled" value="false"/>
<preference name="android-minSdkVersion" value="24"/>
<preference name="BackupWebStorage" value="none"/>
<preference name="SplashMaintainAspectRatio" value="true"/>
<preference name="FadeSplashScreenDuration" value="300"/>
<preference name="SplashShowOnlyFirstTime" value="false"/>
<preference name="SplashScreen" value="screen"/>
<preference name="SplashScreenDelay" value="10000"/>
<preference name="LoadUrlTimeoutValue" value="60000"/>
<preference name="AndroidXEnabled" value="true"/>
<!-- ... -->
<edit-config file="AndroidManifest.xml" mode="merge" target="/manifest/application" xmlns:android="http://schemas.android.com/apk/res/android">
  <application android:usesCleartextTraffic="true"/>
</edit-config>
<config-file target="AndroidManifest.xml" parent="/manifest" xmlns:android="http://schemas.android.com/apk/res/android">
  <uses-permission android:name="android.permission.BLUETOOTH_ADMIN"/>
  <uses-permission android:name="android.permission.INTERNET"/>
</config-file>
```

Please let me know if I should provide any more info. At this moment I'm doing more tests on this issue.

peitschie, 2021-11-20:
I've addressed part of this in https://github.com/don/cordova-plugin-ble-central/pull/845, which will close the connections on app destruction.

The issue with the connections not being closed when Bluetooth is disabled is slightly different. Are you able to confirm how you know the connection is being held open in this instance?

Under normal circumstances, the connect error callback won't be triggered by disabling Bluetooth as it's not the peripheral that has terminated the connection, but the phone itself.

tigohenryschultz, 2022-02-16:
I'm seeing this issue too.

Is there a way to detect the plugin is already connected?

I don't want to have my users have to force close the app.

peitschie, 2022-10-14:
> Is there a way to detect the plugin is already connected?

Just using `ble.isConnected(deviceId, success, failure)` should tell you if you are already connected to the device.
If you just want to try and re-connect, the best approach here is to just call `ble.connect(deviceId...)` on application launch.

@tigohenryschultz if there are still issues around this, feel free to raise a new issue and we can discuss there.