don/cordova-plugin-ble-central #828: BLE scan observable not notifying on first click on iOS platform (Ionic React Capacitor project)

imranaftab, 2020-12-09:
Hi there, 

I am working on latest Ionic v5 with React and capacitor. I have included BLE package and also added all required permissions. The problem I am facing right now is on iOS platform with BLE scan() method (the same code works perfectly fine on Android).  On iOS, BLE.scan() method doesn't notify the subscribe callback when any peripheral is discovered. It only happens the first time I click the button. When I click the scan button again, it works as expected and retrieves the discovered peripheral. Can someone suggest something? Thanks in advance!

For reference, I have checked-in a sample project with this issue on public repo here:
https://github.com/imranaftab/Ionic-react-capacitor-BLE-scan-issue-

Below is the sample code that causes problem on iOS platform:
```
import React, {useState} from 'react';
import { IonButton, IonContent, IonHeader, IonPage, IonTitle, IonToolbar } from '@ionic/react';
import ExploreContainer from '../components/ExploreContainer';
import './Tab1.css';
import { BLE } from '@ionic-native/ble';

const Tab1: React.FC = () => {
  let devices: string[] = []
  const [response, setResponse] = useState('')
  let scan = ()=>{
    BLE.scan([],4).subscribe((device) =>{
      console.log(device);
      if(device && device.name){
        devices = [...devices, device.name];
        setResponse(devices.reduce((a,b)=> a + ', ' + b ));
      }
      
    });
  };

  return (
    <IonPage>
      <IonHeader>
        <IonToolbar>
          <IonTitle>Tab 1</IonTitle>
        </IonToolbar>
      </IonHeader>
      <IonContent fullscreen>
        <IonHeader collapse="condense">
          <IonToolbar>
            <IonTitle size="large">Tab 1</IonTitle>
          </IonToolbar>
        </IonHeader>
        <IonButton onClick={()=> scan()}>Scan</IonButton>
        <p>{response}</p>
      </IonContent>
    </IonPage>
  );
};

export default Tab1;

```

tijmenvangulik, 2021-01-06:
I have the same problem when I run startScan() on ios13. When I look at the log the first time I try to connect the request is blocked by an error message "can only accept this command while in the powered on state" . The second time that I try to connect it works fine.

2021-01-06 18:20:54.484020+0100 Ergometer space[20308:7717741] startScan
2021-01-06 18:20:54.484117+0100 Ergometer space[20308:7717741] [CoreBluetooth] API MISUSE: <CBCentralManager: 0x283f9df00> **can only accept this command while in the powered on state**
2021-01-06 18:20:54.514358+0100 Ergometer space[20308:7717741] Status of CoreBluetooth central manager changed 5 State powered up and ready (CBCentralManagerStatePoweredOn)

Path-A, 2021-03-05:
Have the same issue with my Ionic-Vue / Capacitor project. The workaround @tijmenvangulik mentioned works, but a proper solution would be nice.

JEricaM, 2021-03-31:
Same issue here. I will try @tijmenvangulik workaround!

delvinwidjaja, 2021-08-19:
Does anyone know if this bug has been fixed or will be fixed?

Path-A, 2021-08-19:
No, it is not fixed. You still have to use a workaround or a different BLE library.

delvinwidjaja, 2021-08-19:
I know this is an issue page, but I really want to know if either my workaround is correct or could possibly produce another error.

Below is my method:
```
public scan() {
    console.log('Start scanning');
    this.devices = []; // clear list

    // Scan for 7 seconds
    this.ble
      .isEnabled()
      .then(() => {
        this.ble.scan([], 7).subscribe(
          (device) => this.onDeviceDiscovered(device),
          (error) => this.onScanError(error)
        );
      })
      .catch((err) => {
        // Try to connect again because bluetooth is disabled
        console.log(`Bluetooth is not enabled. Error: ${err}`);
        this.scan();
      });
  }
```

Basically, on the first click, the catch block will be called and the second attempt to scan will be executed. In this state, the bluetooth will be in the enabled state.

peitschie, 2021-11-15:
I've just encountered this issue recently. It's likely caused by the lazy-loading of the BLE bridge here: https://github.com/imranaftab/Ionic-react-capacitor-BLE-scan-issue-/blob/61d5021df934f1fe7150a257802edbdea89b4baa/ios/App/App/config.xml#L6

This is on my radar to fix in the coming weeks.

graphefruit, 2021-11-20:
Great this one is getting fixed :) Thanks for this. 
Funny enough its not an issue on every iOS device I've encountered on iPhone 8 its working, on iPhone 11 its not working on my side

peitschie, 2021-12-07:
Noting for myself, one attempt which might affect this was previously made at https://github.com/don/cordova-plugin-ble-central/pull/735

That approach is what I'd entertain here, will need to figure out the threading issue described there however.

bcole808, 2025-01-02:
Still experiencing this issue on `"cordova-plugin-ble-central": "1.7.8",` with an iPhone 13 Pro. 

Here's a workaround I'm using, which also handles when bluetooth isn't available:

```js
// Workaround:
function enableBluetooth(max_wait = 5000) {
  return new Promise((resolve, reject) => {
    if (typeof ble === "undefined") {
      return reject();
    }

    const interval = setInterval(() => {
      checkStatus();
    }, 100);

    const timeout = setTimeout(() => {
      clearInterval(interval);
      reject();
    }, max_wait);

    const checkStatus = () => {
      ble.isEnabled(() => {
        clearInterval(interval);
        clearTimeout(timeout);
        resolve();
      });
    };

    checkStatus();
  });
}

// Usage:
enableBluetooth()
  .then(() => {
    ble.startScanWithOptions();
  })
  .catch(() => {
    console.log("Bluetooth was not ready even after waiting for it to initialize");
  });
```