don/cordova-plugin-ble-central #831: Encryption array buffer

Allamprabhu11, 2021-01-12:
Is it possible to encrypt and decrypt the array buffer what we getting from the peripheral device?

Thanks

tellybrown, 2021-01-21:
That seems outside the bounds of this project.  I'm sure there are many encryption plugins out there to use.

peitschie, 2021-11-20:
Hi @Allamprabhu11 

The security of the connection is better handled using Bluetooth pairing or bonding, which will transparently encrypt the traffic between the phone and the peripheral.

Any additional layers of encryption beyond this need to be specifically implemented in the peripheral also, so there is no one-size-fits all solution, unfortunately.

I'll close this out as there's not really a feasible solution that works within the plugin for this need. However, feel free to continue the discussion here if desired.

closed 2021-11-20 by peitschie