don/cordova-plugin-ble-central #834: No notify after write

Missile71, 2021-01-22:
I developed a ionic app to connect to NRF52 chip from Nordic Semiconductor.
on ios it's all ok, but on android can connect to device and write to characteristic, but don't receive notify and no error shown.

Ionic:

   Ionic CLI          : 6.12.3 (C:\Users\Missile\AppData\Roaming\npm\node_modules\@ionic\cli)
   Ionic Framework    : ionic-angular 3.9.9
   @ionic/app-scripts : 3.2.4

Cordova:

   Cordova CLI       : 10.0.0
   Cordova Platforms : android 9.0.0
   Cordova Plugins   : 
cordova-plugin-advanced-http 3.1.0 "Advanced HTTP plugin"
cordova-plugin-androidx-adapter 1.1.3 "cordova-plugin-androidx-adapter"
cordova-plugin-ble-central 1.3.1 "BLE"
cordova-plugin-device 2.0.2 "Device"
cordova-plugin-document-viewer 1.0.0 "SitewaertsDocumentViewer"
cordova-plugin-file-opener2 3.0.5 "File Opener2"
cordova-plugin-file 6.0.2 "File"
cordova-plugin-firebasex 11.0.3 "Google Firebase Plugin"
cordova-plugin-globalization 1.11.0 "Globalization"
cordova-plugin-inappbrowser 4.1.0 "InAppBrowser"
cordova-plugin-ionic-keyboard 2.2.0 "cordova-plugin-ionic-keyboard"
cordova-plugin-ionic-webview 5.0.0 "cordova-plugin-ionic-webview"
cordova-plugin-splashscreen 6.0.0 "Splashscreen"
cordova-plugin-statusbar 2.4.2 "StatusBar"
cordova-plugin-whitelist 1.3.3 "Whitelist"
phonegap-plugin-barcodescanner 8.1.0 "BarcodeScanner"

Utility:

   cordova-res : 0.6.0
   native-run  : 0.2.8

System:

   Android SDK Tools : 26.1.1 (D:\adt-bundle-windows\sdk)
   NodeJS            : v10.13.0 (C:\Program Files\nodejs\node.exe)
   npm               : 6.4.1
   OS                : Windows 10

tellybrown, 2021-01-22:
I am on same version of ionic as you with no problem on Android side.  Only thing i can think of is are you telling it to notify using the read uuid?

peitschie, 2021-11-20:
You'll need to specifically start notifications before doing the write to a characteristic.
Notifications are only returned if the peripheral and characteristic you are writing to replies with this.

I'll close this out for now, but feel free to carry on the discussion here.

closed 2021-11-20 by peitschie