don/cordova-plugin-ble-central #836: Autoconnect stops working after a day or so (iOS - Ionic 5)

KyleJJ, 2021-03-08:
I'm setting up autoConnect for the user's devices when the app is launched using the code below. This works great for about a day and a half, then stops working. According the [docs](https://github.com/don/cordova-plugin-ble-central#autoconnect) autoConnect is not supposed to timeout and should wait forever for the device to come back in range. I'm not sure if I'm doing something wrong, or if this is a bug with the plugin, or perhaps someone has found a workaround?

`
this.ble.scan([], 5).subscribe(device => {});
setTimeout(() => {
  for (let i = 0; i < this.devices.length; i++) {
    this.ble.autoConnect(this.devices[i].id, connected => this.deviceConnected(i), disconnected => {});
  }
}, 5000);
`

Any help is greatly appreciated, thank you.

KyleJJ, 2021-03-10:
I did some more digging and it turns out iOS is killing the app after some time, and although [they do have methods to restore bluetooth from the background](https://developer.apple.com/library/archive/documentation/NetworkingInternetWeb/Conceptual/CoreBluetooth_concepts/CoreBluetoothBackgroundProcessingForIOSApps/PerformingTasksWhileYourAppIsInTheBackground.html#//apple_ref/doc/uid/TP40013257-CH7-SW5), this plugin does not have them implemented. I'm switching to [randdusing's BLE plugin](https://github.com/randdusing/cordova-plugin-bluetoothle) which appears to have those methods implemented.

peitschie, 2022-02-15:
As of 1.4.2, this behaviour is hopefully fixed: https://github.com/don/cordova-plugin-ble-central/pull/888

There is now support for iOS state restoration, which allows the connect call to continue to operate as expected.
Note, when iOS relaunches the application due to a background Bluetooth event, the app will need to call `autoConnect` once again to resume the connection loop. Failure to do so within 10s may result in iOS shutting the app back down again.

More discussion on this behaviour can be found at #717

closed 2022-02-15 by peitschie