don/cordova-plugin-ble-central #837: The plugin doesnt work in background mode

Absant94, 2021-03-17:
Hi!

Im using this plugin with : 
[https://github.com/katzer/cordova-plugin-background-mode](https://github.com/katzer/cordova-plugin-background-mode)

The problem is that this works without problems and catch the events.

But when I move it to the background the problem appears. The BLE conection is still working, but this stop to catch events.

This line suscribe to the service and catch the events: 

```
showServices(data){
  this.services=[];
  this.services = data.characteristics;
  for(var i = 0;i<this.services.length;i++){
    
    if(this.services[i].service.includes("1523") && this.services[i].characteristic.includes("1524")){
   
      this.ble.startNotification(data.id,this.services[i].service,this.services[i].characteristic).subscribe(  ()=>this.botonPulsado(),() =>this.showError("FAIL") )
      
    }
  }
}
```

How can I solve it?

Thanks!

benplain, 2021-08-24:
I am also interested in this on iOS. I am able to get background mode working on Android using forks of https://github.com/katzer/cordova-plugin-background-mode.
I imagine that it would be a common requirement to monitor BLE devices while in the background.

peitschie, 2022-02-15:
This now works on iOS (with some caveats) as of 1.4.2.
See the `BLUETOOTH_RESTORE_STATE` discussion in the [README](https://github.com/don/cordova-plugin-ble-central#ios)

The PR that landed this was https://github.com/don/cordova-plugin-ble-central/pull/888

One caveat that exists has been documented in this issue: https://github.com/don/cordova-plugin-ble-central/issues/891

I'll close this issue as done, but feel free to re-open if there's some use-case that's been missed with these changes.

closed 2022-02-15 by peitschie