don/cordova-plugin-ble-central #838: Icant send data to HM10

00alkskodi00, 2021-03-21:
Hi

I can scan, connect, disconnect, but after connect i cant send the data to HM10, id and uuids are 100 percent good

here is my part of code:

var string = "sn";

var device_id = "34:14:B5:41:B9:62"; // mac aresa bluetooth zariadenia BLE
var serviceUUID = "0000ffe0-0000-1000-8000-00805f9b34fb";
var service_uuid = "0000ffe1-0000-1000-8000-00805f9b34fb";
var characteristic_uuid = "0000ffe1-0000-1000-8000-00805f9b34fb";

// ASCII only
function stringToBytes(string) {
    var array = new Uint8Array(string.length);
    for (var i = 0, l = string.length; i < l; i++) {
        array[i] = string.charCodeAt(i);
    }
    return array.buffer;
}

function sendData() {
    var data = stringToBytes(string);
    console.log("odoslal som string: " + data)
    var success = function () {
        console.log("data sended");
    };

    var failure = function () {
        console.log("Failed writing data to the redbear hardware");
    };
    ble.writeWithoutResponse(device_id, service_uuid, characteristic_uuid, data, success, failure);
}

00alkskodi00, 2021-03-21:
in RedBear example working all with my id and uuids

coloz, 2021-03-21:
I tested the HM10 / HM11, and they can communicate normally with the plugin

00alkskodi00, 2021-03-21:
Yes i know but i am begginer in programming 

Here is my complete js file

/*
 * Licensed to the Apache Software Foundation (ASF) under one
 * or more contributor license agreements.  See the NOTICE file
 * distributed with this work for additional information
 * regarding copyright ownership.  The ASF licenses this file
 * to you under the Apache License, Version 2.0 (the
 * "License"); you may not use this file except in compliance
 * with the License.  You may obtain a copy of the License at
 *
 * http://www.apache.org/licenses/LICENSE-2.0
 *
 * Unless required by applicable law or agreed to in writing,
 * software distributed under the License is distributed on an
 * "AS IS" BASIS, WITHOUT WARRANTIES OR CONDITIONS OF ANY
 * KIND, either express or implied.  See the License for the
 * specific language governing permissions and limitations
 * under the License.
 */

'use strict';

// Wait for the deviceready event before using any of Cordova's device APIs.
// See https://cordova.apache.org/docs/en/latest/cordova/events/events.html#deviceready
document.addEventListener('deviceready', onDeviceReady, false);

function onDeviceReady() {
    // Cordova is now initialized. Have fun!

    console.log('Running cordova-' + cordova.platformId + '@' + cordova.version);
    document.getElementById('deviceready').classList.add('ready');
    refreshButton.addEventListener('touchstart', sendData, false);
    pripojbutton.addEventListener('touchstart', pripoj, false);
    odpojbutton.addEventListener('touchstart', odpoj, false);

    skenuj()
    checkBLE();
    checkGPS();
   // pripoj();
    //siPripojeny()
}

var string = "sn";

var device_id = "34:14:B5:41:B9:62"; // mac aresa bluetooth zariadenia BLE
var serviceUUID = "0000ffe0-0000-1000-8000-00805f9b34fb";
var service_uuid = "0000ffe1-0000-1000-8000-00805f9b34fb";
var characteristic_uuid = "0000ffe1-0000-1000-8000-00805f9b34fb";

// funkcia skenuj
function skenuj() {
    ble.startScan([], function (device) {
        console.log(JSON.stringify(device));
    }, console.log(""));

    setTimeout(ble.stopScan,
        5000,
        function () { console.log("Scan complete"); },
        function () { console.log("stopScan failed"); }
    );
}

function pripoj() {

    var onError = function () {
        console.log("nemozem sa pripojit");
    };

    var onConnect = function () {
        console.log("pripojeny");
    };

    ble.connect(device_id, onConnect, onError);

}

function onData(data) { // data received from Arduino
    console.log(data);
}

// ASCII only
function stringToBytes(string) {
    var array = new Uint8Array(string.length);
    for (var i = 0, l = string.length; i < l; i++) {
        array[i] = string.charCodeAt(i);
    }
    return array.buffer;
}

// odosli data
function sendData() {
    var data = stringToBytes(string);
    console.log("odoslal som string: " + data)
    var success = function () {
        console.log("data sended");
    };

    var failure = function () {
        console.log("Failed writing data to the redbear hardware");
    };
    ble.writeWithoutResponse(device_id, service_uuid, characteristic_uuid, data, success, failure);
}

function checkBLE() {
    ble.isEnabled(
        function () {
            console.log("Bluetooth is enabled");
        },
        function () {
            console.log("Bluetooth is *not* enabled");
            alert("Zapni bluetooth");
        }
    );
}

function checkGPS() {
    ble.isLocationEnabled(
        function () {
            console.log("location services are enabled");
        },
        function () {
            console.log("location services are *not* enabled");
            alert("Zapni lokaizáciu");
        }
    );
}

function odpoj() {
    console.log("funkcia odpoj");
    ble.disconnect(device_id, success, failure);
    var success = function () {
        console.log("odpojeny");
    };
    var failure = function () {
        console.log("failure");
    };
}

function siPripojeny() {
    setTimeout(function () {
        ble.isConnected(
            device_id,
            function () {
                console.log("Peripheral is connected");
            },
            function () {
                console.log("Peripheral is *not* connected");
            }
        );
    }, 1000);
}

peitschie, 2021-11-20:
@00alkskodi00 It's a bit hard to read that code snippet there to tell what's going on.

What you've got here looks reasonable (presuming you touch the buttons in the right order).
If you haven't already, I'd suggest trying to do this manually using nrf Connect (https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp) as a test, to ensure your aurduino is correctly configured.

peitschie, 2022-07-07:
Closing due to inactivity. Please re-open if this is still an issue for you.

closed 2022-07-07 by peitschie