don/cordova-plugin-ble-central #840: Send file via ble

Darkmagister, 2021-04-01:
Hello,
sorry but this i don't think it's an issue but i didn't know where to write,
i'm writing an app with ionic that communicate with an esp32(arduino framework) and i would like to add the updatefirmware function, so i need to send the firmware from the app to the esp32, but i'm not really sure how to achieve that, as the firmware it's around 1mb do you have any suggestion on how to send the file ?
i have to split the file and do multiple write ?

thanks a lot.

peitschie, 2021-11-15:
Hi @Darkmagister, I'm not aware of how arduino firmware can be pushed via Bluetooth. A quick google turns up things like https://learn.sparkfun.com/tutorials/esp32-ota-updates-over-ble-from-a-react-web-application/all , which suggests it's possible.

This isn't on my radar to implement however, as I don't really work with that stack. However, if additional native code is required for Android or iOS to support this use-case, patches are definitely welcome.

don, 2021-12-21:
Your ESP32 needs to have an update firmware service or characterisic. This could be something custom you write or you could reuse the Nordic UART service. For a more complicated example look at the [Nordic DFU](https://github.com/don/cordova-plugin-ble-central#nordic-dfu) implementation.

For the simple example, create an updateFirmwareCharacteristic on the ESP32 that has write properties. Make this characteristic as large as the ESP32 and your phone support, say 200 bytes. On the phone, split your 1MB firmware into 200 byte chunks and write each chunk to the updateFirmwareCharacteristic. You could add a firmwareChecksum characteristic too. Once the phone is done writing the firmware, it can send a checksum to the firmwareChecksum characteristic so the ESP32 can verify the upload and install the firmware.

peitschie, 2022-11-11:
Just spring-cleaning a bit. The plugin supports all the low level things required to create such a service in javascript. I can't see any improvements or additional features needed to the core plugin itself 🙂 

Feel free to re-open if there's a specific need here that you believe isn't filled.

closed 2022-11-11 by peitschie