don/cordova-plugin-ble-central #848: Using autoconnect function always asks user to pair

digoocorrea, 2021-04-29:
Hi there, I'm currently working on an ionic capacitor app that needs to connect automatically with a ble device.

The goal is sending and receiving bluetooth info and commands to the ble device, which is a car alarm central that controls a few actions like lock and unlock the doors accordingly to the RSSI values on from the device. 

So, I'm using the autoconnect function to auto-connect with the device as soon the device is close to range and all works perfectly in Android and iOS. The only issue is that everytime the device reachs the ble range and the ble plugin try connecting, the system prompts the user to accept pairing with the device again.

I need it to work without that pairing prompting everytime. I tried to pairing the device previous to initialize the app but it didn't worked as well. So is it a way to make this autoconnection work without prompting the pairing acceptance everytime it is disconnected?

The issue occurs on Android versions 9 and 10 and iOS.

Thank you in advance.

digoocorrea, 2021-05-05:
UPDATE

I've changed my ble hardware interface to ask for a PIN code. I guessed the connection could be considered secure by the OS and prevent prompting the user to pair again. Didn't work as well.

Any toughts? Anyone? Thank you!

peitschie, 2021-11-20:
@digoocorrea this is often caused by the peripheral (not the phone) failing to remember the exchanged security keys. In Bluetooth, there are 3 different levels, being unsecured, paired (temporary key exchange) and bonded (permanent key exchange).

Bonding is what causes the Pairing dialogs to show up in Android. In this situation, the peripheral should remember the keys and re-use them every time the phone reconnects. If the peripheral forgets the keys every time, Android will assume it's not the same peripheral, and will prompt the user to rebond again.

closed 2023-04-21 by peitschie