don/cordova-plugin-ble-central #850: Ionic app with regular Background BLE tasks

giowild, 2021-05-03:
Hi all, does someone have already implemented an ionic+cordova only app(no extra Java code) able to perform regularly a background task with a BLE connection? 

In particular I would like to perform a BLE connection, connect, fetch and disconnect, once every 15 minutes, in bakground, to a device with a known service UUID, with initial scanForServiceUUID after OS reboot or after Bluetooth adapter state change.
I know there are background cordova plugins (https://github.com/katzer/cordova-plugin-background-mode) but these do not seem to perform background tasks REGULARLY and continously across phone restarts.

It would be great to know if anyone has already successfully done it and if additional Java code has been required. 

Thanks a lot!

dariosalvi78, 2021-05-03:
can't be done with cordova currently. See https://github.com/apache/cordova/issues/120

giowild, 2021-05-03:
> can't be done with cordova currently. See [apache/cordova#120](https://github.com/apache/cordova/issues/120)

Just found this:
https://github.com/transistorsoft/cordova-plugin-background-fetch

Did you already know that?

dariosalvi78, 2021-05-03:
this looks like exactly what I was suggesting in that issue! What I am not sure is if the app has access to plugins though. Worth trying!

(and if you try it, let us know please)

giowild, 2021-05-03:
Of course, I'm going to try that and also ask developer if plugin can fit this kind of use cases. He seems a very kind person...
Stay tuned

peitschie, 2024-03-20:
There's an open question on Capacitor's background runner about whether plugin support is a possibility: https://github.com/ionic-team/capacitor-background-runner/issues/24