don/cordova-plugin-ble-central #852: ble.write on ios return null (on Android returns OK)

JEricaM, 2021-05-03:
I'm using this plugin inside a ionic + stencil app
https://github.com/don/cordova-plugin-ble-central

I'm send a write command to a characteristic that support WRITE.
If I call ble.write on Android, the callback returns me OK
If I call ble.write on iOS, the callback returns me NULL.
in both cases, I can see on my peripheral device that the command has been sent correctly.

I connect to the device like this:

      connectToDevice(device){
    var toast = new utils();
    let tclass;
    let tmessage;

    console.log('----connectToDevice----');
    BLE.connect(device.id).subscribe( 
        peripheralData => {
          this.device = peripheralData;
          let connectBtn = document.querySelector('#connectBtn') as any;
          console.log('device connected: ', this.device);
        },
        error => {
          console.log('connect error: ', error);
        }
    );
  }

And then, send the command with this code:

    async sendCommandNew(id){
        let noteOk;
        let noteError;
    
        let data;
        data = this.stringToBytes(this.commandToSend);
        let timerId = setInterval(() => {
          BLE.write(id, "49535343-fe7d-4ae5-8fa9-9fafd205e455", "49535343-1e4d-4bd9-ba61-23c647249616", data).then((res) => {
            console.log('res: ', res);
           
          }).catch(() => {
            console.log('res no');
          }
          )
        }, 1000);
        setTimeout(() => { clearInterval(timerId); console.log('stopped'); }, 5000);
      }

This is the characteristic

[image]

I've checked issues and saw various people with this problem but I didn't figure it out how to solve this.

peitschie, 2021-11-20:
Hi @JEricaM 

Are you able to clarify what you are using this return value for?

From the plugin code, the "OK" you are receiving here is not actually from the peripheral, but merely indicates that the write succeeded. If the write had failed, you would end up in your `catch()` block there.

You can see here in code to see that there is no actual data being returned from the write.

https://github.com/don/cordova-plugin-ble-central/blob/master/src/android/Peripheral.java#L452

Just as an aside, the Bluetooth spec doesn't actually allow the peripheral to return data directly on a write. In most cases, data is returned from the peripheral using an indication or notification.

Techwarehut, 2021-11-20:
`Hi @peitschie, I tried doing a ble.read after the write and all I got is the device name. Am I missing a step? Same thing for notify, callback function is called accurately but when I stringify the response - I get a blank string?

Any help is much appreciated!`

peitschie, 2021-11-24:
@Techwarehut the `read` is trying to read a specific characteristic. I can't answer as to whether you're doing the wrong or right thing here, as that's entirely up to what the Bluetooth device you're talking requires you to do.

A `read` is not the same as receiving an `indicate` or `notify` message... again, it's all down to what the  Bluetooth device's on BLE interface documentation says you should be doing.

closed 2022-06-29 by peitschie