don/cordova-plugin-ble-central #853: BLE encrypted characteristics + Huawei phones

Doomkyn, 2021-05-05:
Hello, 
I recently found some annoying behaviour when trying to access encrypted characteristics through a Huawei phone.
I'll explain my tests:

SAMSUNG PHONE:
- Connecting and Reading non-encrypted characteristics => ok
- Reading encrypted characteristics open a system dialog which asks for a PIN to bond devices, once done => ok

HUAWEI PHONE:
- Connecting and Reading non-encrypted characteristics => ok
- Reading encrypted characteristics => fails with error 137

I also tried using setPin and/or createBond, nothing works on Huawei

If I use the NRF Connect app to read an encrypted characteristics, the system correctly prompts for bonding with PIN.
Once I bond this way, I can read encrypted characteristics in my Ionic App

Has anybody experienced this behaviour? Any tips?

Thanks.

peitschie, 2021-11-20:
Hi @Doomkyn 

I haven't experienced this issue with my own Huawei phone. Are you able to provide more detail about the Android version and model? It's also worth checking if your phone has a previous pairing with the peripheral, as in some versions of Android having an existing key that's no longer valid can cause some hiccups like this.

Failing all that, can you confirm if you see this issue with the nrf Connect app (https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp)?

peitschie, 2023-02-19:
Closing this out due to inactivity. Feel free to re-open if the issue is still occurring.

closed 2023-02-19 by peitschie