don/cordova-plugin-ble-central #854: Android: "peripheral disconnected" if disconnect and then connect

JEricaM, 2021-05-11:
Hi
It's an Android problem.
I need, after BLE.write, to disconnect the device and connect it again, so I call this function after BLE.write done successfully 

```
reconnectDevice(device) {
    BLE.disconnect(device.id).then(() => {
      BLE.connect(device.id).subscribe(
        peripheralData => {
          this.device = peripheralData;
          localStorage.setItem('device', JSON.stringify(this.device));
          error => {
            console.log('Error: ', error);
          }
        }
      );
    });
  }
```

The problem is that BLE.connect always returns the failure callback saying "peripheral disconnected" and I don't understand why. 
as the doc says 
https://github.com/don/cordova-plugin-ble-central#connect
`The disconnect callback is not called when the application calls ble.disconnect. The disconnect callback is how your app knows the peripheral inintiated a disconnect.`
I saw other issues about this but I saw that the problem was solved with a commit. Btw I still have this problem and I've added to my project the last plugin version.
Maybe I call the two functions wrong?

zackcornelius, 2021-06-24:
By default on Android, the BLE supervision timeout is 20 seconds (but Android will accept shorter values if the device sends it in a connection update). On iOS, this default appears to be 720ms.

This means that Android doesn't complete the disconnect process until the supervisory timeout has passed, so for 20 seconds after you call disconnect, connect will return Peripheral Disconnected - really stating that the peripheral is actively disconnecting.

If you have control of the peripheral side, you can send a connection update request with a shorter supervisory timeout (Apple recommends 2-6 seconds in their BLE development guide) and reduce this time.

JEricaM, 2021-06-28:
Thank you so much for the information.
Now it make sense. 
I can't have control of the peripheral unfortunately but at least I have an answer for this question.
Thank you again