don/cordova-plugin-ble-central #859: How to detect bad PIN when connecting with secure characteristic?

HankLloydRight, 2021-07-27:
I have an ESP32 project where I have configured the BLE server to ask for a PIN before connecting securely/encrypted.  

Everything works great with BLE-CENTRAL when the user enters the proper PIN (the mobile device pops up the PIN dialog box).  Once the user enters the correct PIN, I can send/receive encrypted data normally.  

But I do not know how to detect in BLE-CENTRAL when the user enters a wrong PIN. I've tried the onError(reason) with ble.connect() but that doesn't work.  I don't know where or how to capture this error condition.

Other mobile BLE apps return the error messages: "_Authentication is insufficient_" and "_Encryption is insufficient_" when a user enters the wrong PIN.  These two messages are common across several different BLE apps. so I think these two messages are coming from the BLE core on the phone.  How can I get these messages back with BLE-CENTRAL when a user does not enter the correct PIN and notify the user?

Thank you.

woodz-, 2021-09-25:
If your solution points to the thing which is called bonding, you could have a look onto [this PR](https://github.com/peitschie/cordova-plugin-ble-central/commit/afdfef875ee53f2d00b7b27bd3d2c572808567c9). Bonding isn' t officially supported for this library.

peitschie, 2024-03-20:
This feature was implemented a while back and released in 1.6.3

closed 2024-03-20 by peitschie