don/cordova-plugin-ble-central #861: ble.write truncate data packet to MTU in Android only

MDB80, 2021-08-12:
Hi guys, 

I wonder if anyone can solve this strange behavior.

I use ESP32 and cordova BLE central.
In android I have set the MTU to 495 bytes.
In ios it's already like that by default.

I've noted that if I send a packet with 800bytes:

- in ios I receive two packets with 495bytes and 305bytes = 800bytes  ; ok perfect!
- in android, instead, I receive two packets with 495bytes and 105bytes = 600bytes ???

200bytes lost: where did they go? truncate?? loses?

Interestingly, the maximum buffer that ESP32 uses is 600bytes. Could this have something to do with it?
But I wonder why with an Iphone this problem not exists.

My guess is that the IOS library correctly splits the data according to the MTU, while the Android library does not;
or IOS and Android have completely different protocols.

Does anyone know anything about this?

Thank you for you attention.

MDB80, 2021-08-12:
I found something [here](https://stackoverflow.com/questions/48741196/android-ble-client-only-returns-600-byte-of-data-in-oncharacteristicread).

For anyone who comes across this post also wondering why Android seems to return only 600 bytes for long GATT characteristics like this question is asking about, it all comes down to how Bluedroid (Android's Bluetooth stack) implements their GATT Client and how its out of spec. In my case, I was using an ESP32-based IoT device as my GATT Server and Android (SDK 24) for the GATT Client.

According the spec (Bluetooth Core 4.2; Vol 3, Part F: 3.2.9), the maximum size for a characteristic value (inherited from ATT's attribute value) is 512 bytes. However, for some reason, Bluedroid does not attempt to enforce this requirement, instead decided upon a maximum size of 600; which can be seen if you dive into the Bluedroid source and find the macro GATT_MAX_ATTR_LEN which is set to 600 (stack/include/gatt_api.h:125). Since in my case (and yours it seems) I was implementing the read request response code, I did not see to enforce the 512 byte limit on reads for characteristics either.

Now, its important to realize just how it seems Bluedroid reads characteristics and how that relates to both the MTU size, the maximum size of a read (should be 512, but is 600 for Bluedroid) and how to handle data longer than that maximum size. The MTU size is the largest packet size on the ATT level you can use. So, for each call to BluetoothGatt.readCharacteristic, you may be sending one or more read requests to the server depending on if Bluedroid thinks the characteristic size exceeds the MTU size. On a low level, Bluedroid will first send a ATT Read Request (0x0a) and if the packet is MTU bytes in length, it will follow up with a ATT Read Blob Request (0x0c) with the offset set to the MTU size. It will continue to send ATT Read Blob Requests until either the ATT Read Blob Response is less than MTU bytes in length or until the maximum characteristic size is reached (ie, 600 for Bluedroid). Its important to note that if the MTU size is not a perfect multiple of 600 for data longer than 600 bytes, the remaining bytes will be discarded (as Bluedroid never actually expects to read 600 bytes since it thinks the GATT Server will enforce the 512 byte limit on characteristic sizes). So, if your data exceeds the 600 byte limit (or 512 limit for safety), you should expect to call BluetoothGatt.readCharacteristic multiple times. Heres a simple example for reading lots of data on the Android side (sorry Im not using bleno so cannot give you the code to fix that side), it relies on first sending the length of the data as a unsigned 32-bit integer and then reading out the data with repeated calls to BluetoothGatt.readCharacteristic if the data is longer than 600 bytes:

peitschie, 2021-11-15:
Hi @MDB80. It sounds like you got to the bottom of your question ok there in there end.

Generally speaking, the maximum value you can read or write should not exceed the MTU that's been negotiated.

Though many stacks (mobile and embedded) support payloads larger than the MTU, it's not consistent. Once you send payloads larger than the MTU, you are doing Long Reads and Long Writes and need to be very careful about checking support for this feature is present and implemented in both mobile and embedded device.

I'll close this out, as it's not clear to me what changes or improvements could be made to the plugin with regards to this handling. However, feel free to continue the discussion here.

closed 2021-11-15 by peitschie