don/cordova-plugin-ble-central #862: BLE scan method is not returning devices. Using ionic 5 ble plugin

mrc-prasad5, 2021-08-15:
I am using ionic 5.4.16 version to develop a bluetooth poc. Installed ble-central plugin as per documentation. But the plugin is not working. Below is my code:

app.module.ts:

import { NgModule } from '@angular/core';
import { BrowserModule } from '@angular/platform-browser';
import { RouteReuseStrategy } from '@angular/router';
import { IonicModule, IonicRouteStrategy } from '@ionic/angular';
import { BLE } from '@ionic-native/ble/ngx';
import { AppComponent } from './app.component';
import { AppRoutingModule } from './app-routing.module';
import { Geolocation } from '@ionic-native/geolocation/ngx';
import { Diagnostic } from '@ionic-native/diagnostic/ngx';
import { LocationAccuracy } from '@ionic-native/location-accuracy/ngx';

@NgModule({
  declarations: [AppComponent],
  entryComponents: [],
  imports: [BrowserModule, IonicModule.forRoot(), AppRoutingModule],
  providers: [BLE,Geolocation, Diagnostic, LocationAccuracy, { provide: RouteReuseStrategy, useClass: IonicRouteStrategy }],
  bootstrap: [AppComponent],
})
export class AppModule {}

home.page.html:

<ion-header [translucent]="true">
  <ion-toolbar>
    <ion-title>
      Bluetooth Demo
    </ion-title>
  </ion-toolbar>
</ion-header>

<ion-content [fullscreen]="true">
  <ion-button (click)="scan()">
    <ion-icon name="refresh-circle-outline"></ion-icon>&nbsp;
    Refresh Bluetooth Devices
  </ion-button>
  <ion-row>
    <ion-col>
        <ion-item *ngFor="let device of devices;let i = index">
          <ion-label>{{device.name || 'Unnamed'}}</ion-label>
          <ion-label>{{device.id}}</ion-label>
          <ion-label>RSSI: {{device.rssi}}</ion-label>
        </ion-item>
    </ion-col>
  </ion-row>
</ion-content>

home.page.ts:

import { Component, NgZone, OnInit } from '@angular/core';
import { BLE } from '@ionic-native/ble/ngx';
import { AlertController, ToastController, Platform, LoadingController } from '@ionic/angular';
import { Geolocation } from '@ionic-native/geolocation/ngx';
import { Diagnostic } from '@ionic-native/diagnostic/ngx';
import { LocationAccuracy } from '@ionic-native/location-accuracy/ngx';

declare var cordova;
@Component({
  selector: 'app-home',
  templateUrl: 'home.page.html',
  styleUrls: ['home.page.scss'],
})

export class HomePage implements OnInit {

  devices: any = [];
  statusMessage: string;

  constructor(
    private toastCtrl: ToastController,
    private ble: BLE,
    private platform: Platform,
    private loading: LoadingController,
    private geolocation: Geolocation,
    private diagnostic: Diagnostic,
    private locationAccuracy: LocationAccuracy,
    private ngZone: NgZone) {
  }

  ionViewWillEnter() {
    
  }

  ngOnInit(){
    this.platform.ready().then(() => {
      this.diagnostic.isLocationEnabled().then(enable => {
        if (enable) {
          console.log('location enabled');
          this.scan();
        } else {
          this.locationAccuracy.request(this.locationAccuracy.REQUEST_PRIORITY_HIGH_ACCURACY).then(() => {
            console.log('location has to be enabled');
            this.scan();
          }).catch(error => {
            console.log('Error requesting location permissions ', JSON.stringify(error));
          });
        }
      }).catch(error => {
        console.log('Location service error ', JSON.stringify(error));
      });
    })
  }

  scan() {
    this.geolocation.getCurrentPosition().then(resp => {
      console.log('location captured');
      this.ble.enable().then(() => {
        this.loading.create({ message: 'Scanning...' }).then(loading => {
          loading.present();
          console.log('Scanning...');
          this.ble.scan([], 5).subscribe(device => {
            console.log(JSON.stringify(device));
            this.onDeviceDiscovered(device);
            loading.dismiss();
          }, error => console.log('error ', JSON.stringify(error)));
        });
      });
    }, error => {
      console.log('Error scanning devices');
    });
  }

  onDeviceDiscovered(device) {
    console.log('Discovered ' + JSON.stringify(device, null, 2));
    this.ngZone.run(() => {
      this.devices.push(device);
    });
  }

  // If location permission is denied, you'll end up here
  async scanError(error) {
    this.setStatus('Error ' + error);
    let toast = await this.toastCtrl.create({
      message: 'Error scanning for Bluetooth low energy devices',
      position: 'middle',
      duration: 5000
    });
    toast.present();
  }

  setStatus(message) {
    console.log(message);
    this.ngZone.run(() => {
      this.statusMessage = message;
    });
  }
}

Please share a possible working solution. You help is appreciated.

Thanks

peitschie, 2021-11-20:
Hi @mrc-prasad5 

Are you able to confirm which platforms this affects? Also, do you know what your cordova-plugin-ble-central configuration is? (Either in `package.json` or in `config.xml`)?

Thanks.

gspeach, 2022-02-02:
I can confirm this is happening on my PC. I am running an apache cordova app on android emulator and its unable to find an esp32 device I'm running with BLE. However, on my PC and my Android phone I can run a scan and see the device.

peitschie, 2022-06-14:
Closing due to inactivity. It's hard to guess what's going on here, unfortunately.

closed 2022-06-14 by peitschie