don/cordova-plugin-ble-central #864: Is there a max nr of messages within a specific time on 1 characteristic

I-Connect, 2021-09-02:
On connect our esp32 sends all sensor values (18 with a total size of around 125 bytes for all messages) via separate notifications on 1 characteristic.

Only 15 are processed in the app. If I change the order of the messages sent I receive still only the first 15.
Adding a delay of 200 mS between sending the messages on the ESP32 does not change it, still only 15 messages arrive in app 
Is this queued somewhere with a limit? 

When a sensor value changes on the esp32 I send the new value, this does arrive at the app for all 18 sensors. (some time has passed then since connect)

Any idea what can be causing this (esp32 or cordova side) and how to fix?

thx!

peitschie, 2021-11-15:
Hi @I-Connect,

There are no limits within the plugin that might affect this that I am aware of. It's possible the esp32 Bluetooth firmware has limitations around this value (this is pretty common in embedded stacks). The other possible consideration, are you specifically using Notifications or Indications? In Bluetooth, only Indications are guaranteed delivery, while Notifications are delivered when possible.

Interesting that the 200ms delay here doesn't seem to help however. Do you have any other phones to test this out with?

don, 2021-12-21:
It's possible you're writing the new values to the characteristic before the ESP can send the notification. It's possible your phone can't handle all the notifications.

1) try using indications instead of notifications
2) increase the time between writes to 1 second or longer then debug with LightBlue or NFR Connect

I'd consider changing the BLE service to have separate characteristics for each sensor value.

closed 2023-02-19 by peitschie