don/cordova-plugin-ble-central #866: Detect Bluetooth autorisation for IOS 13.1 +

dufrenb, 2021-09-06:
Since IOS 13.0, Bluetooth requires user permission but first they forgot to give us a way to check for autorisation without asking for them.
In IOS 13.1, Apple provide us à way to do that. See https://developer.apple.com/documentation/corebluetooth/cbmanager/3377595-authorization?changes=latest_major.

Is it possible to implement this ?

peitschie, 2021-11-15:
Hi @dufrenb , this is somewhat non-trivial to implement in the plugin currently. The `CBManager` is initialised as soon as the plugin is loaded, which automatically prompts the authorisation request.

See https://github.com/don/cordova-plugin-ble-central/blob/f5569e48262da901ee3b82b81ade3c73c5418f8b/src/ios/BLECentralPlugin.m#L41

I'm happy to consider patches to address this however, if a solution can be found that doesn't require too much code there to be refactored.

peitschie, 2022-02-25:
Just noting, this is related to these discussions
* https://github.com/don/cordova-plugin-ble-central/pull/769
* https://github.com/don/cordova-plugin-ble-central/pull/735
* https://github.com/don/cordova-plugin-ble-central/issues/739#issuecomment-619149028