don/cordova-plugin-ble-central #872: startScan fails silently if location services are disabled

peitschie, 2021-11-15:
As per https://github.com/don/cordova-plugin-ble-central/blob/f5569e48262da901ee3b82b81ade3c73c5418f8b/src/android/BLECentralPlugin.java#L712

This should fail the scan callback so the calling app understands that the scanning is not in progress.

peitschie, 2022-02-15:
Seems there's a bit of history around this behaviour

* https://github.com/don/cordova-plugin-ble-central/pull/633
* https://github.com/don/cordova-plugin-ble-central/issues/607
* https://github.com/don/cordova-plugin-ble-central/pull/595

Any solution here will need to figure out how to determine whether the scan might work or not...

sam-higgs, 2022-05-20:
Any updates on this?

peitschie, 2022-05-23:
Hi @sam-higgs 

Nothing specific yet. The linked discussions above suggest there are some phones out there that are allowed to scan despite the location service being disabled, so I'm reluctant to force it one way or another.

In practice, this is something the application itself can do by checking `ble.isLocationEnabled` manually before starting a scan: https://github.com/don/cordova-plugin-ble-central#islocationenabled

But... it's still on my radar!