don/cordova-plugin-ble-central #874: installation

ASHBAW, 2021-11-16:
Sorry for putting this here. I am a newb and was wanting to know how to install version 1.4.0?

If I try:
`cordova plugin add cordova-plugin-ble-central`

it installs version 1.3.1.

If I try:
`cordova plugin add cordova-plugin-ble-central@1.4.0`

I get the following:

```
Failed to fetch plugin cordova-plugin-ble-central@1.4.0 via registry.
Probably this is either a connection problem, or plugin spec is incorrect.
Check your connection and plugin name/version/URL.
CordovaError: Error: npm: Command failed with exit code 1 Error output:
npm ERR! code ETARGET
npm ERR! notarget No matching version found for cordova-plugin-ble-central@1.4.0.
npm ERR! notarget In most cases you or one of your dependencies are requesting
npm ERR! notarget a package version that doesn't exist.
```

I am not sure what I am doing wrong?

Thanks for any help.

peitschie, 2021-11-17:
Hello! The 1.4.0 release isn't yet published to NPM. I'll post back here once it is.

peitschie, 2021-11-19:
Hi @ASHBAW , 1.4.0 has been released! Please yell out if you have any further issues with this.

closed 2021-11-19 by peitschie