don/cordova-plugin-ble-central #877: Unable to use requestConnectionPriority

jlaverdure-talis, 2021-11-22:
Hello,

I meet a problem with my application that is far too slow to receive packages from the device. In classic, the data transfer has a 0.45 seconds duration whereas the Low Energy transfer reaches 1,25 seconds per package with the same amount of data.

I read about the requestConnectionPriority function but I'm unable to use it since it doesn't exist on the latest @ionic-native/ble module.

Could you please explain me how should I proceed ?

Thank you,

peitschie, 2021-11-22:
Hi @jlaverdure-talis 

I'm not that familiar with who or how the ionic native ble wrapper is maintained, unfortunately, so don't have any particular insights.

The function in this plugin here is used like this:
```
ble.requestConnectionPriority(device_id, priority, () => console.log("success"), (error) => console.error("failed", error))
```

Where priority is one of:

- https://developer.android.com/reference/android/bluetooth/BluetoothGatt#CONNECTION_PRIORITY_BALANCED
- https://developer.android.com/reference/android/bluetooth/BluetoothGatt#CONNECTION_PRIORITY_HIGH
- https://developer.android.com/reference/android/bluetooth/BluetoothGatt#CONNECTION_PRIORITY_LOW_POWER

huiigary, 2022-08-23:
Hi, I have same issue. 

In the index.d.ts file of "/node_modules/@ionic-native/ble/ngx" there is no reference to requestConnectionPriority. Do we need to upgrade Cordova to resolve this issue?

    "cordova-plugin-ble-central": "^1.4.1",
    "@ionic-native/ble": "^5.31.1",

peitschie, 2022-08-23:
@huiigary it appears that this method has not been exposed in the ionic wrapper: https://github.com/danielsogl/awesome-cordova-plugins/blob/master/src/%40awesome-cordova-plugins/plugins/ble/index.ts

I'd suggest perhaps raising an issue against that repository to have these new methods exposed.

Closing this out as it's purely an ionic plugin issue, which I don't maintain.

peitschie, 2022-08-23:
Seems like there was an issue raised that never got actioned for some reason: https://github.com/danielsogl/awesome-cordova-plugins/issues/3737

There was even a PR request that was raised, but it looks like it was closed without ever being merged. Perhaps you could rebase that PR and resubmit, in case the repository owners are interested?

PR was up at https://github.com/danielsogl/awesome-cordova-plugins/pull/3738

closed 2022-08-23 by peitschie