don/cordova-plugin-ble-central #880: Android 11 Permissions problem.

monkeytronics, 2021-12-12:
Hi there, 

I've been using this plugin with no issues on Android 10 phones. However, it seems that when I try it on a newer android 11 phone, it does not work. I notice that BLE permissions are never asked for in Android 11. I wonder why this would be? Is this something other users have encountered? I'm working in ionic 5.4.9 and using cordova...

    "@ionic-native/ble": "^5.24.0",
    "cordova-android": "^7.1.4",
    "cordova-plugin-ble-central": "^1.3.1",

Thanks, 

Al.

peitschie, 2021-12-13:
Hi @monkeytronics 

Are you able to confirm which target SDK version the application was compiled with?

monkeytronics, 2021-12-13:
Hey @peitschie,

Targeting Android 11, API 30. Minor update, I'm working on an ionic project, and I managed to work around this by installing the android permissions cordova plugin, and requesting permission :

```
if (this.platform.is('android')) {
      let res = await this.androidPermissions.requestPermission(this.androidPermissions.PERMISSION.ACCESS_COARSE_LOCATION);
      console.log('req ACCESS_COARSE_LOCATION = ', res);
    }

 await this.ble.isEnabled().then(
      ret => console.log('bluetooth enabled'),
      err => {
        console.log('bluetooth not enabled. Ask user...')
        this.ble.enable().then(ret => {
          console.log('bluetooth switch on : ', ret); 
        });
      });

    this.scan();
```

peitschie, 2021-12-13:
Interesting! Theoretically, coarse location should only be needed if you're targeting Android 28 or lower: https://developer.android.com/guide/topics/connectivity/bluetooth/permissions#declare-android11-or-lower

Is this on every phone, or is it a specific manufacturer showing the issue?

monkeytronics, 2021-12-15:
More updates. First I updated the ble-central to the latest version. Then I tested the code above on a phone using Android 11 GO and was able to get it to work. Another phone (not mine), on Android 11 proper, can see the permission pop up, but the scan does not run - as before. Are you able to replicate this one at your end? To debug this, I'll need to get develop access to the phone, which I feel a bit funny about asking people! But if I do, I'll share the logs and / or fixes, if I find any. 

Do you have any theories or things you'd like me to try out while I'm at it??

peitschie, 2021-12-15:
The main thing I would confirm @monkeytronics is that the user has enabled Location Services. If the user has disabled Location, the scan will silently fail, though you'll still get the permissions pop-up. There's an open defect around this https://github.com/don/cordova-plugin-ble-central/issues/872.

You could confirm this in code by checking the response for [isLocationEnabled](https://github.com/don/cordova-plugin-ble-central#islocationenabled) before starting the scan.

peitschie, 2022-06-29:
Closing due to no feedback for a while. I am not able to reproduce this problem with any of the Android 11 phones I have access to.

However, please feel free to re-open this if the issue is still occurring!

closed 2022-06-29 by peitschie