don/cordova-plugin-ble-central #881: Cant find Jolimark Bluetooth Printers

ebofi, 2021-12-16:
Jolimark tp510 is a widely used blutooth receipt printer and this plugin cant find the said printer when we initiate the scan command. SO do you have any turnaround for this.

peitschie, 2021-12-17:
Are you able to see the printer if you scan with another Bluetooth app such as [nrf Connect](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp)?

HankLloydRight, 2021-12-17:
I would suspect the printer is a regular Bluetooth device and not BLE.

ebofi, 2021-12-17:
The manufacturer told it's Bluetooth 4.0 BLE. any advice

HankLloydRight, 2021-12-17:
Then do what peitshie suggested and use other BLE apps to test. There are many apps for Android and iOS.

ebofi, 2022-01-07:
Hi,

I tried with BLE Scanner and nrf connect but the printer is not listing, but on regular phone Bluetooth scan the printer shows.

@peitschie @HankLloydRight @don @timkim 

Anyone please advise me how to connect this jolymark to a cordova project( we use Monaca for building)

peitschie, 2022-01-21:
Hi @ebofi 

Based on the [printer specifications](https://www.jolimark.com/download/manual_en/tp510.pdf), this printer might not a Bluetooth Low Energy device. Versions of Bluetooth earlier than 4.0 are referred to as Bluetooth Classic, and won't be visible when using the Bluetooth LE scanner.

[image]

[image]

Perhaps have a look at something like https://www.npmjs.com/package/cordova-plugin-bluetooth-classic-serial-port and see if that is better suited to your needs?

I'll close this out for now, but feel free to continue the conversation here.

closed 2022-01-21 by peitschie