don/cordova-plugin-ble-central #890: How to Use Xpress Streaming Service

alexanderrudnik, 2022-02-14:
Hello all!

I have a small and working python script that imitates BLE devices. Also, I bought a Silicon Labs BGX220P board. I am installing that script on the board and testing it through [BGX Commander](https://play.google.com/store/apps/details?id=com.silabs.bgxcommander&hl=ru&gl=US). And I see that this is working. I am using [Xpress streaming service](https://docs.silabs.com/gecko-os/1/bgx/latest/ble-services#xpress-streaming-service)

But when I am trying to create a simple ionic app using `cordova-plugin-ble-central` and that board to use Xpress streaming service there, I am facing some difficulties: I can send a command to RX Characteristics and can see that board catches it and sends me a response. But I cannot see that response in my app :(

Here is what I am doing: 
```
  BLE.connect(id).subscribe(data => {
        this.peripheral = data
        this.setStatus('Connected to ' + name)
        
        // Once connected, I am starting listening to TX Characteristics according to the Xpress Streaming Service Docs
        BLE.startNotification(id, this.service, this.tx_char).subscribe(value =>
        {
          this.setStatus('get notification: ' + bytesToString(value) + '----' + JSON.stringify(value) )
        },
          err => this.setStatus("Error " + err),
          (res) => this.setStatus('subscribed: ' + res)
        )
      })
```

Then on some test button I do: 
```
BLE.write(this.device.id, this.service, this.rx_char, stringToBytes('{"cmd": 1}\n'))
```

I see that `write` function works as expected and the board catches `{"cmd": 1}`. But no notification from TX Characteristics occur. What am I doing wrong?

Huawei P20 Pro, Android 10

Thank you

peitschie, 2022-02-15:
Hi @alexanderrudnik 

I can't see anything obviously wrong with what you're doing there with regards to the setup of the notifications and writing.
What response are you expecting to receive for that command?

Is the command handling custom, out of interest? I noticed the command you're forming there doesn't really look much like the format I'd expect based off https://docs.silabs.com/gecko-os/1/bgx/latest/commands...

alexanderrudnik, 2022-02-15:
Hi @peitschie 

thank you for your attention!

It's a custom command, I am expecting the following: (screenshot from BGX Commander)
[image]

peitschie, 2022-02-15:
Is it possible that you need to set the Stream Mode? I notice there's a toggle there in the app which is set to STREAM...

[image]

alexanderrudnik, 2022-02-15:
Stream mode is enabled.

OK, I found something: it works on iPhone 12 and Xiaomi Android 9!
So your plugin is working!

The issue is in the Huawei smartphone or Android 10+ OS. 
I know that the plugin adds required permissions for Android 10+ itself, but still: can it be an Android 10 OS issue? I think it's a Huawei issue as I found an opinion on the Internet that Huawei is known for its BLE bugs... I'm going to dive into that problem.

So, as we found out that the plugin does work, we can close the issue.

Thank you for help!

peitschie, 2022-02-15:
Fantastic. Glad to hear you managed to get things working.

In theory, the only permissions difference on Android 10 is for scanning... once you are connected, there should be no impact from those changes.

If you're looking for ways to dig further, I'd suggest running the app under Android Studio and using the Logcat console to see whether there's anything erroneous being reported there.

Best of luck!