don/cordova-plugin-ble-central #891: iOS: Unable retrieve initial scan/notification event that caused state restoration

peitschie, 2022-02-15:
Related to #888 and #717

The current state restore implementation has no way to retrieve the exact scan or notification event that caused the iOS app to be restored.

State restoration occurs as soon as the manager is initialised, and iOS will then immediately invoke the relevant delegate callback (e.g., `centralManager:didDiscoverPeripheral:`) with the event that caused the app to be woken up. An app has no opportunity to register a callback handler at this point however, so this event will be missed.

Somehow, the plugin needs to understand that state restoration is active and allow the calling application to retrieve any scan or notification events that occur during the initialise phase.

adelin-mcbsoft, 2022-02-15:
Hi @peitschie ,

Sorry for not answering in the other thread, but I was just waiting to find the right time to test the new implementation before I come back with any feedback.

I suppose that having that object available at the application start should be 'good enough' to do the BLE routine even though if we don't have an event -- but as said, didn't get the time to test it yet, although I wanted to give a 'shout out' at least, to let you know that I'm aware that you made the change and I'm looking forward to test it out soon.

Thank you once again for all the time invested in this and I'll come back with feedback ASAP,
Have a good one,
A.

peitschie, 2022-02-15:
No dramas at all @adelin-mcbsoft ! Can appreciate that time is always hard to find.

Feel free to test when you have a moment... any feedback is welcome for certain! I'm using this myself in a few applications, so expect some tweaks will flush out of those as well.