don/cordova-plugin-ble-central #894: isConnected return true even after turning bluetooth on/off from Android phone

tigohenryschultz, 2022-02-16:
We are turning bluetooth on/off in the middle of a connection and receiving a write error. When we get a write error we check if the device isConnected, which is returning true but it should not be.

If the user disables bluetooth it should disconnect all sessions.

peitschie, 2022-02-17:
I've been able to confirm this issue. Will dig into it in the next few days!

peitschie, 2022-02-17:
I've pushed up a pre-release build if you're able to confirm that this fixes the issue for you: https://www.npmjs.com/package/cordova-plugin-ble-central/v/1.4.3-alpha.1

peitschie, 2022-02-18:
@tigohenryschultz are you able to confirm whether the 1.4.3-alpha.1 release fixes this issue for you?

tigohenryschultz, 2022-02-18:
I will try tomorrow, I was having trouble building the Android app today with it.

tigohenryschultz, 2022-02-18:
@peitschie 

I get this error in AndroidStudio when trying to build/deploy to my phone:

```
I:\projects\tigosmart\android\capacitor-cordova-android-plugins\src\main\java\com\megster\cordova\ble\central\BLECentralPlugin.java:526: error: cannot find symbol
                    int connectedState = bluetoothManager.getConnectionState(peripheral.getDevice(), BluetoothProfile.GATT);
                                                                                                     ^
  symbol:   variable BluetoothProfile
  location: class BLECentralPlugin
```

peitschie, 2022-02-21:
I've pushed 1.4.4-alpha.2, which should fix the compile issue. Apologies for that hiccup!

closed 2022-03-14 by peitschie