don/cordova-plugin-ble-central #898: Accu Chek Mobile with Bluetooth adaptator

BenLaKnet, 2022-02-19:
Hi,

I try to read datas from Accu Chek Mobile with Bluetooth adaptator but I do not know what is the values for :

service_uuid, 
characteristic_uuid

Could you help me please.

Many thanks

peitschie, 2022-02-21:
Perhaps https://github.com/nativescript-community/ble/issues/134 might contain some hints.

There appears to be a Bluetooth standard for these, that you might have some luck finding in https://www.bluetooth.com/specifications/specs/ and by searching for glucose references in https://btprodspecificationrefs.blob.core.windows.net/assigned-values/16-bit%20UUID%20Numbers%20Document.pdf

peitschie, 2022-02-21:
An application like [nrf Connect](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp) may help you discover the details for yourself here. I'd suggest install that and see if you can figure out those identifiers from there 🙂

peitschie, 2022-02-21:
Closing this for now as there's no plugin related impact. However, feel free to continue the discussion here!

closed 2022-02-21 by peitschie

BenLaKnet, 2022-02-22:
@peitschie Many thanks.

I found these 2 values :

service_uuid		= "0x1808";
characteristic_uuid = "0x2A51";

How must I inject values for reading characteristic.

peitschie, 2022-02-22:
You would pass those into the `ble.read` instruction. For example:

```
ble.read(device_uuid, service_uuid, characteristic_uuid, 
  (data) => console.log("data received:", data),
  (error) => console.error("Read failed:", error));
```

BenLaKnet, 2022-02-22:
I do that, but it returns read failed.
It is necessary to transform values into something ?

peitschie, 2022-02-22:
What's the exact read error you get? I'd suggest logging the `failure` variable somewhere.

I presume you have connected to the device already?

BenLaKnet, 2022-02-23:
This is my code :

	service_uuid		= "1808";
	characteristic_uuid = "2A51";

	
	ble.read(localStorage.getItem('ACMid'), service_uuid, characteristic_uuid,
		function(data){
			addMyContenu("Hooray we have data : " + JSON.stringify(data), 1);
			console.log("Hooray we have data : " + JSON.stringify(data));
		},
		function(failure){
			addMyContenu("Failed to read characteristic from device." + JSON.stringify(failure), 1);
			console.log("Failed to read characteristic from device." + JSON.stringify(failure));
		}
	);

And now I do not have error message, but an empty string for results "{}"

peitschie, 2022-02-23:
`data` there is usually an `ArrayBuffer`. Try doing `new Uint8Array(data)`, and see if that gives you any extra data?

BenLaKnet, 2022-02-23:
I do that

ble.read(localStorage.getItem('ACMid'), service_uuid, characteristic_uuid,
		function(data){
			
			data = new Uint8Array(data);
			
			addMyContenu("Hooray we have data : " + JSON.stringify( data ), 1);			
			console.log("Hooray we have data : " + JSON.stringify( data ));
			
		},
		function(failure){
			addMyContenu("Failed to read characteristic from device. " + JSON.stringify(failure), 1);
			console.log("Failed to read characteristic from device. " + JSON.stringify(failure));
		}
	);

it returns : "Hooray we have data : {"0":32,"1":1}"

BenLaKnet, 2022-02-23:
And now, I have : 

"Hooray we have data : {"0":288}"

peitschie, 2022-02-23:
🎉 

Glad to hear it @BenLaKnet

BenLaKnet, 2022-02-24:
OK, Many thanks @peitschie 

I obtain different values : manufacturer, model name, serial number.
I'm searching for glucose...

BenLaKnet, 2022-02-25:
@peitschie 
I do not understand how to obtain glucose measurement.
I am trying to read GATT documentation.

peitschie, 2022-02-25:
@BenLaKnet What you're trying to do here (*write* to the characteristic rather than read it) sounds a bit funny to me... but I'm not at all familiar with the glucose measurement specification.

I'd highly recommend using [nrf Connect](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp) or a similar tool to get familiar with how to read and work with the data.

Reading the GATT spec itself is not necessarily the best approach to learning Bluetooth. I'd suggest googling a bit more as there seems to be some [stackoverflow answers](https://stackoverflow.com/a/63340809) on the topic, as well as simpler demo code snippets available when you google something like "bluetooth glucose measurement read dev"

BenLaKnet, 2022-02-27:
@peitschie I am using nrf Connect to discover and control services and characteristics.

But I read that I need to write something to :  
service_uuid : "1808",
characteristic_uuid : "2A52",

for obtaining all datas measurement or last or first : 
https://stackoverflow.com/questions/45029498/reading-from-a-notify-characteristic-ionic-bluetooth

BenLaKnet, 2022-02-28:
@peitschie  I do that and it is running, now I have to decrypt value : 

var databuffer = new Uint8Array(2);
	databuffer[0] = 0x01;
	databuffer[1] = 0x01;
		
	console.log("startnotif");
	
	ble.startNotification(localStorage.getItem('ACMid'), "1808", "2A18", 
		function(datanotif){
			
			int8data = new Uint8Array(datanotif);			
			console.log("Hooray we have data notif : " + JSON.stringify( int8data ));			
				
		},
		function(failure){
			addMyContenu("Failed notif from device. " + JSON.stringify(failure), 1);
			console.log("Failed notif characteristic from device. " + JSON.stringify(failure));
		});		
	
	console.log("startwrite");
	
	ble.write(localStorage.getItem('ACMid'), "1808", "2A52", databuffer.buffer, 
		function(data){
			console.log("Hooray we have data write : " + JSON.stringify( data ));				
		}, 
		function(failure){
			addMyContenu("Failed write from device. " + JSON.stringify(failure), 1);
			console.log("Failed write characteristic from device. " + JSON.stringify(failure));
		});

BenLaKnet, 2022-03-01:
@peitschie Many thanks, I convert correctly values : 

int8data = new Uint8Array(datanotif);			
						
			id = int8data[1] & 255;
			id |= (int8data[2] & 255) << 8;
			
			val = int8data[12];
					
			annee = int8data[3] & 255;
			annee |= (int8data[4] & 255) << 8;			
			
			mois = addZero(int8data[5]);
			jour = addZero(int8data[6]);
			
			h = addZero(int8data[7]);
			m = addZero(int8data[8]);
			s = addZero(int8data[9]);
			
			datehms = annee + "-" + mois + "-" + jour + "T" + h +":"+m+":"+s;

peitschie, 2022-03-01:
Sounds like you've made great progress there @BenLaKnet ... nice job!

peitschie, 2022-03-01:
If you weren't already aware of it, the DataView class might make the decoding step there a bit easier to do: https://devdocs.io/javascript/global_objects/dataview/getuint16

BenLaKnet, 2022-03-02:
@peitschie many thanks,

I will try that.