don/cordova-plugin-ble-central #899: Waiting for a response

davidtan29, 2022-02-23:
I'm creating an app for a proprietary piece of hardware. The device supports writeWithoutResponse only. I'm able to connect to the device, start notifications, write to the characteristic and get a reply, so all is good there.

However, different commands evoke different kinds of replies. Most are just one group of characters, but some send back 4 or 5 lines of responses. I'm capturing them all and logging to the console, and storing the responses in a database table.

The problem is i trigger the next write command off the previous. Even with a delay, my commands are stepping on each other. In other words, i'll send the next write command while I'm still receiving notifications from the previous. I think I get all the data, but it goes into the log and the database out of order. Ideally there is an end character in all the responses and I want to wait for that until I send the next command. But I'd like it to be smooth.

Is there a way with startNotifications to wait for a complete response before moving to the next command? I'm probably missing some simple logic but I've been messing with this hardware for longer than I even want to think about and my brain has gone to mush.

peitschie, 2022-02-23:
Hi @davidtan29 

The key question is how do you know when all the data has been received? There's no way for `startNotification` to tell whether multiple notification packets are part of the same message or not, so it's going to be in your app and peripheral to figure this out 🙂

Common approaches I've seen here are:
1. Have the first notification include a byte length prefix. E.g., if the command response is going to be 300 bytes, and the longest is 1024 bytes, the first notification would be (in bytes): `[0x01, 0x2C, .... data ...]`
2. Have a special notification to indicate end-of-message (which you'd already mentioned)
3. Have an idle timeout. E.g., when no notifications are received for a certain length of time (e.g., 200ms), assume the packet is complete and continue with the next command.

In essence, all of these are executed with some variation of the following (presuming you have `async/await` syntax... can be done with straight promises if needed though).

```
// Somewhere before any commands are sent:
let commandResponse = [];
let bytesReceived = 0;
let lastPacketReceived;

ble.startNotifications(device_id, service_id, characteristic_id,
  (data) => { 
    commandResponse.push(data);
    lastPacketReceived = Date.now();
    bytesReceived += data.byteLength;
  },
  (err) => console.error("notification error", err)
);

// 1. reset buffer before each command
commandResponse = [];

// 2. send the command
ble.writeWithoutResponse(device_id, service_id, characteristic_id, ...data);

// 3. decide how the packet is complete
... if length based, something like ...
// bytesToUint16 impl is something you can google
const packetLength = bytesToUint16(commandResponse[0], commandResponse[1]);
while (bytesReceived < packetLength) {
 await delay(some timeout); // delay can be found in the prex library if you want
}

... if time-based ...
await delay(inital response timeout);
while (Date.now() - lastPacketReceived < IDLE_TIMEOUT_MS) {
 await delay(IDLE_TIMEOUT_MS);
}

// 4. Now repeat steps 1-3 for each command
```

Does this help at all?

davidtan29, 2022-02-25:
This is super helpful! Thank you so much for the thoughtful reply. I'm going to work on implementing the idle timeout - i think that will work best based on what I've seen from the software. I'll reply back with any follow-up questions. Again, thank you so much!

peitschie, 2022-03-14:
I'll close this for now, but feel free to continue the discussion here or re-open if you hit any issues!

closed 2022-03-14 by peitschie