don/cordova-plugin-ble-central #901: Exception after turning off bluetooth while scanning devices

AngeNew, 2022-02-24:
Hi, I have a problem, I am using ble-central plugin in ionic app, but After turning bluetooth while bluetooth scanning devices, I have this error:
E/AndroidRuntime: FATAL EXCEPTION: JavaBridge
Process: io.ionic.starter, PID: 8409
java.lang.IllegalStateException: BT Adapter is not turned ON
at android.bluetooth.le.BluetoothLeUtils.checkAdapterStateOn(BluetoothLeUtils.java:136)
at android.bluetooth.le.BluetoothLeScanner.stopScan(BluetoothLeScanner.java:256)
at com.megster.cordova.ble.central.BLECentralPlugin$5.run(BLECentralPlugin.java:887)
at android.os.Handler.handleCallback(Handler.java:808)
at android.os.Handler.dispatchMessage(Handler.java:101)
at android.os.Looper.loop(Looper.java:166)
at android.os.HandlerThread.run(HandlerThread.java:65)
How can I trouble this problem??
Thanks

peitschie, 2022-02-25:
I can confirm this is a bug in the BLE plugin itself. It seems the `stopScan` call ought to be gated behind a check to see if the Bluetooth adapter is still enabled

https://github.com/don/cordova-plugin-ble-central/blob/3bdab048bcac41c48b8fe636adeef7c31704f1bb/src/android/BLECentralPlugin.java#L887

AngeNew, 2022-02-25:
Thanks for your response, app crash after this error. Do I avoid turning off bluetooth?? is it only solution??

peitschie, 2022-02-25:
Keeping the Bluetooth on would certainly stop the crash happening. Out of interest, why are you turning off Bluetooth during the scanning?

It would be great to fix the issue in the plugin though. Patches welcome if you have the time to help... otherwise I'll keep it on my radar for a fix when I get a chance 🙂

AngeNew, 2022-02-25:
I turning off bluetooth for a possible senior user case :), If I have more time in some weekend, I try to trouble this problem

peitschie, 2022-02-25:
Ah yes... test cases always cause troubles 😆 

The right fix here for the plugin would be two parts:

1. Add a check [here](https://github.com/don/cordova-plugin-ble-central/blob/0732380e27772a08df2c6ce6447b4cab901ad4d2/src/android/BLECentralPlugin.java#L993) to check whether `bluetoothAdapter.getState()` is still [STATE_ON](https://developer.android.com/reference/android/bluetooth/BluetoothAdapter#getState()) before the `stopScan` call is made
2. The second bit would be to surround the [stopScan](https://github.com/don/cordova-plugin-ble-central/blob/0732380e27772a08df2c6ce6447b4cab901ad4d2/src/android/BLECentralPlugin.java#L994) call in a try/catch (no log needed) so that any errors resulting from calling stop scan don't terminate the application.

AngeNew, 2022-02-25:
For me is also right way. 
In this weekend, I try to modify, but I don't assure :)

commit 98ba3e3d (2022-10-14): Android: Prevent various exceptions on stopScan #901 #871

Fixes #901 and #871
commit b3f384f7 (2022-10-14): Android: Prevent various exceptions on stopScan #901 #871

Fixes #901 and #871
commit 26a142dc (2022-10-18): Android: Prevent various exceptions on stopScan #901 #871

Fixes #901 and #871
closed 2022-10-18 by peitschie