don/cordova-plugin-ble-central #904: Plugin stops when in background mode

MarcinDl, 2022-03-03:
Hello folks,

scanning stops emitting beacons when screen is off or is the app is in the background 

installing background mode - https://ionicframework.com/docs/native/background-mode - didn't help.  

does anyone has similar problem and was able to solve it?

any tips, tricks, hack, workarounds or solutions are most welcome!

Best regards, 
Marcin

peitschie, 2022-03-03:
@MarcinDl significant improvements were made in this area very recently in v1.4.2

Have a look at https://github.com/don/cordova-plugin-ble-central#background-scanning-and-notifications-on-ios , as you'll need to set the `BLUETOOTH_RESTORE_STATE` variable to `true` in order for this feature to activate.

peitschie, 2022-03-03:
Just to clarify, is this Android or iOS or both?

MarcinDl, 2022-03-04:
till now on Android. As I read your tip was for iOS, wasn't it? 

But do you have any super simple code that should work in background with continuous scanning? I've been struggling with it for days with no results.

peitschie, 2022-03-08:
@MarcinDl out of interest, do you have `ACCESS_BACKGROUND_LOCATION` set to true? E.g., https://github.com/don/cordova-plugin-ble-central#android

In general however, the decision to pause the scan sits in how Android's battery optimisation behaves. There's no reliable way to keep the app executing on a wide variety of phones, and most solutions require native Android code (e.g., foreground services, background services, work manager, etc.).

I'll leave this open for now however as this is definitely an area I'd like improved someday...

peitschie, 2022-06-29:
@MarcinDl I've recently found that the Cordova webview will pause execution roughly 5mins after being put into the background.

Perhaps have a look at https://bitbucket.org/TheBosZ/cordova-plugin-run-in-background/src/master/, specifically `disableWebViewOptimizations` to see if this is of any help.

peitschie, 2023-02-19:
I'm closing this out. Feel free to continue the discussion here or re-open if there's still remaining issues unaddressed in your question.

closed 2023-02-19 by peitschie