don/cordova-plugin-ble-central #909: Ios error stopStateNotifications

AngeNew, 2022-03-24:
Hi, I use in my app startStateNotifications in android works fine, in Ios works but after call startStateNotifications calls stopStateNotifications without use in my code, 
Xcode return me this:
2022-03-24 09:43:27.008398+0100 App[657:59780] Status of CoreBluetooth central manager changed 4 State BLE powered off (CBCentralManagerStatePoweredOff)
2022-03-24 09:43:27.008498+0100 App[657:59780] Report Bluetooth state "off" on callback BLE751184174
To Native Cordova ->  BLE stopStateNotifications INVALID ["options": []]
To Native Cordova ->  BLE stopStateNotifications INVALID ["options": []]

this happens only after use first startStateNotifications in app, after first time works fine.
How can I trouble??

peitschie, 2022-03-25:
@AngeNew is this using the ionic wrapper? If so, are you able to see if you experience the same problem using the plugin directly without going via the wrapper?

AngeNew, 2022-03-25:
No, I am not using ionic wrapper, I use plugin in constructor of my component, but I see one thing, if I use startStateNotificatons in IonViewDidEnter return error, instead if I use in button  this method, id is valid.

AngeNew, 2022-03-25:
I use this component, in two different scenarios, and I see in one of this, all works fine, and in the other scenario doesn't work

peitschie, 2022-03-27:
Interesting. Are you able to provide more details about the two scenarios? I can't think of an obvious reason why the state notifications would stop like this. Is there any chance you can provide a reproduction source repo I could test with?

AngeNew, 2022-03-29:
it is as if the error came from the fact that at the first start it sees the bluetooth off, and therefore it cannot stop the status, because if you see when it gives me the 2 invalid calls, the bluetooth status is off, but the bluetooth was on , while I was doing the test, but at the first start he sees it turned off

AngeNew, 2022-03-30:
Another thing that I want to question is this, what is better approach to test my component bluetooth??
and what is coverage test percentage in your plugin?? because, I use plugin in app that respect CE certification

peitschie, 2022-04-11:
> Another thing that I want to question is this, what is better approach practically to test my bluetooth component?? and what is coverage test percentage in your plugin?? because, I use plugin in app that respects CE certification

The short answer is there is very minimal coverage within this plugin. CE certification generally requires the hardware and BLE firmware to be considered also, which is all well outside the bounds of what this plugin does. Unfortunately, there's no standard approach I can suggest here as it all depends on the regulations applicable for your industry.

peitschie, 2022-04-11:
@AngeNew based on the above code snippet, there seems to be quite a bit of complexity in how you're wiring up the Ionic wrapper with your own code.

The crux of the issue however looks to me like it's caused by calling `stopStateNotifications` when showing the BLE disabled dialog. If you disable state notifications, how does your app detect when the user re-enables the BLE again?

[image]

I'd suggest try removing all the `stopStateNotification` calls except for the one done when leaving the view, and see if that improves the behaviour.

It's very expected that the BLE adapter may report as initially off on iOS, due to how the BLE adapter initialises on first access.

AngeNew, 2022-04-11:
thanks, I understand for test question and I try delete all stopStateNotification  calls and reorganize states in my app, thanks also for this

peitschie, 2022-04-12:
My pleasure! I'll close this for now, but feel free to re-open if you need more help with this @AngeNew 🙂